gnab/rtl8812au #29: Raspberry Pi Alfa Network AWUS036ac no ad-hoc

nordmoen, 2015-01-07:
Using the driver in this library together with the AWUS036AC containing the "Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter"(lsusb output) I could not get Ad-hoc working. I did not test any other functionality and I'm not well versed in C so I did not dig that hard through the source. However, by using the driver found on Alfa networks site: http://www.alfa.com.tw/download_show.php?combo_0=&combo_1=&combo_2=&keyword=awus036ac&verify=7rrkpx&x=0&y=0 (driver direct link: https://www.dropbox.com/s/6uruc5og6xw1877/rtl8812au_linux-master.tar.gz?dl=0) and adding the if statement about Raspberry Pi compilation from this project, I got the device running and Ad-hoc mode working. Just thought I would throw that out there in case there are any significant changes or additions in Alfa network's driver.

duchenpaul, 2015-03-07:
how did u compile the driver, i have got lot problem while compiling,same Adapter as urs

hitam4450, 2015-08-27:
Any feedback from @nordmoen  about your solution, yet  ??
regards, hiatm4450