gnab/rtl8812au #35: Issue with aircrack

gothman59, 2015-02-21:
Hi, 

I'm trying to use aircrack on ubuntu with this driver but i have an error so if anybody can help it would be great :)

Here's my error : 

$sudo airodump-ng -t wep wlan0
ioctl(SIOCSIWMODE) failed: Invalid argument

ARP linktype is set to 1 (Ethernet) - expected ARPHRD_IEEE80211,
ARPHRD_IEEE80211_FULL or ARPHRD_IEEE80211_PRISM instead.  Make
sure RFMON is enabled: run 'airmon-ng start wlan0 <#>'
Sysfs injection support was not found either.

Thanks

hanswolff, 2015-02-22:
Did you try to run "airmon-ng start wlan0"? This should create a new monitor adapter which you can use with airodump-ng.

hanswolff, 2015-02-22:
Also it could be that there is no monitor mode support in that driver / chipset.

gothman59, 2015-02-22:
Yes i did " airmon-ng start wlan0 " and it answer me "monitor mode enabled" but there is no new adapter.
And when i do " airmon-ng start wlan0 " i get this : 

Found 4 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to kill (some of) them!

PID	Name
1102	NetworkManager
3505	wpa_supplicant
3725	avahi-daemon
3726	avahi-daemon

but i can't kill any of them, i know the wpa_supplicant come from the driver but i don't know what are the avahi-daemon one.

Thanks for you answer.

tz1, 2015-02-22:
These are part of standard Linux networking.  I normally just become root
and do "killall avahi-daemon; killall NetworkManager; killall
wpa_supplicant".  Avahi is mdns/zeroconf/rondezvous/bonjour.  They can
reconfig the device.

Also, iwconfig will show monitor mode or not.

Monitor should work - I don't remember since I use a different device
(alfa), but injection or other special features might not.

gothman59, 2015-02-22:
Thanks for your answer, i tried to do all of this in root but i still can't kill all the process i also tried " iwconfig wlan0 mode monitor " but it says "permission refused" while i'm in root ... so i guess there's no monitor mode with this chipset.

tz1, 2015-02-22:
Do "ps alx |less" and look for a parent that keeps respawning the
processes.  You may need to kill that too/first