gnab/rtl8812au #38: Syntax error: word unexpected when compiling for Raspberry Pi 2

CRTX, 2015-03-06:
I tried to compile this driver for my Raspberry 2 but I run into the following error.

```
$ make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.18.8-v7+/build M=/home/pi/Desktop/rtl8812au  modules
make[1]: Entering directory '/lib/modules/3.18.8-v7+/build'
  CC [M]  /home/pi/Desktop/rtl8812au/core/rtw_cmd.o
./scripts/recordmcount: 1: ./scripts/recordmcount: Syntax error: word unexpected (expecting ")")
scripts/Makefile.build:257: recipe for target '/home/pi/Desktop/rtl8812au/core/rtw_cmd.o' failed
make[2]: *** [/home/pi/Desktop/rtl8812au/core/rtw_cmd.o] Error 2
Makefile:1381: recipe for target '_module_/home/pi/Desktop/rtl8812au' failed
make[1]: *** [_module_/home/pi/Desktop/rtl8812au] Error 2
make[1]: Leaving directory '/lib/modules/3.18.8-v7+/build'
Makefile:1051: recipe for target 'modules' failed
make: *** [modules] Error 2
```

*/lib/modules/3.18.8-v7+build/scripts/Makefile.build*

```
# Built-in and composite module parts
$(obj)/%.o: $(src)/%.c $(recordmcount_source) FORCE
        $(call cmd,force_checksrc)              #<--------------LINE 257
        $(call if_changed_rule,cc_o_c)
```

The code for line 257 looks about right... anyone know what the problem might be?

SamSolve, 2015-03-11:
I've had the following issue for Raspberry Pi 2 on 3.18.7 kernel:

pi@raspberrypi ~/rtl8812au/rtl8812au $ uname -a
Linux raspberrypi 3.18.7-v7+ #756 SMP PREEMPT Wed Feb 18 16:14:51 GMT 2015 armv7l GNU/Linux
pi@raspberrypi ~/rtl8812au/rtl8812au $ ls
clean  core  hal  ifcfg-wlan0  include  Kconfig  Makefile  os_dep  README.md  runwpa  wlan0dhcp
pi@raspberrypi ~/rtl8812au/rtl8812au $ vim.tiny Makefile
pi@raspberrypi ~/rtl8812au/rtl8812au $ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.18.7-v7+/build M=/home/pi/rtl8812au/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux'
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_cmd.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_security.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_debug.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_io.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_mlme.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_vht.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_rf.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_recv.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_ap.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_xmit.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_p2p.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_tdls.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_br_ext.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_iol.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/rtw_sreset.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/os_dep/osdep_service.o
  CC [M]  /home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.o
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c: In function ‘create_proc_read_entry’:
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:328:2: warning: initialization from incompatible pointer type [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:328:2: warning: (near initialization for ‘fops.read’) [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c: In function ‘create_proc_read_write_entry’:
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:339:3: warning: initialization from incompatible pointer type [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:339:3: warning: (near initialization for ‘fops.read’) [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:341:2: warning: initialization from incompatible pointer type [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:341:2: warning: (near initialization for ‘fops.write’) [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c: In function ‘rtw_proc_init_one’:
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:389:3: warning: passing argument 4 of ‘create_proc_read_entry’ from incompatible pointer type [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:321:38: note: expected ‘ssize_t (**)(struct file *, char *, ssize_t,  loff_t *)’ but argument is of type ‘int (*)(char *, char **, off_t,  int,  int *, void *)’
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:438:8: warning: passing argument 4 of ‘create_proc_read_write_entry’ from incompatible pointer type [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:333:38: note: expected ‘ssize_t (**)(struct file *, char *, ssize_t,  loff_t *)’ but argument is of type ‘int (*)(char *, char **, off_t,  int,  int *, void *)’
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:438:8: warning: passing argument 6 of ‘create_proc_read_write_entry’ from incompatible pointer type [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:333:38: note: expected ‘ssize_t (**)(struct file *, const char *, ssize_t,  loff_t *)’ but argument is of type ‘int (*)(struct file *, const char *, long unsigned int,  void *)’
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:445:8: warning: passing argument 4 of ‘create_proc_read_write_entry’ from incompatible pointer type [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:333:38: note: expected ‘ssize_t (**)(struct file *, char *, ssize_t,  loff_t *)’ but argument is of type ‘int (*)(char *, char **, off_t,  int,  int *, void *)’
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:445:8: warning: passing argument 6 of ‘create_proc_read_write_entry’ from incompatible pointer type [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:333:38: note: expected ‘ssize_t (**)(struct file *, const char *, ssize_t,  loff_t *)’ but argument is of type ‘int (*)(struct file *, const char *, long unsigned int,  void *)’
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:452:8: warning: passing argument 4 of ‘create_proc_read_entry’ from incompatible pointer type [enabled by default]
/home/pi/rtl8812au/rtl8812au/os_dep/linux/os_intfs.c:321:38: note: expected ‘ssize_t (**)(struct file *, char *, ssize_t,  loff_t *)’ but argument is of type ‘int (*)(char *, char **, off_t,  int,  int *, void *)’
[... cut ...]
pi@raspberrypi ~/rtl8812au/rtl8812au $ ls
8812au.ko     8812au.mod.o  clean  hal          include  Makefile       Module.symvers  README.md  wlan0dhcp
8812au.mod.c  8812au.o      core   ifcfg-wlan0  Kconfig  modules.order  os_dep          runwpa
pi@raspberrypi ~/rtl8812au/rtl8812au $ sudo insmod 8812au.ko
Error: could not insert module 8812au.ko: Invalid module format
pi@raspberrypi ~/rtl8812au/rtl8812au $ uname -a
Linux raspberrypi 3.18.7-v7+ #756 SMP PREEMPT Wed Feb 18 16:14:51 GMT 2015 armv7l GNU/Linux

igorpecovnik, 2015-03-11:
Can you compile some other wireless drivers with this kernel for test? This kind of error you can get if something vital is missing (i don't recall what I need to add to make it work in my case) in kernel regarding wireless. Just test some other wireless drivers and if you will always get "Invalid module format" than you need to rebuild kernel ...

itay13-meet, 2015-03-17:
Did you find any solution for this issue? I really want to use my Edimax AC600 but I can't because I didn't find any way to fix the compiling issue.

Thanks

SamSolve, 2015-03-17:
Going to be looking into this over the next few days.

daryltucker, 2015-04-16:
Looking forward to it, @SamSolve

amerckado, 2015-05-11:
Did any of you find a solution? just bought Edimax AC600 and can't believe it's useless... :(

itay13-meet, 2015-05-14:
This post solved my issue:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=66499&p=721331&hilit=8812au#p721331