gnab/rtl8812au #40: Doesnt work on a Fedora 21 **i686**

aviallon, 2015-04-10:
Cant build on my i686 computer. I've tried make -i, I've had more text on my screen, but at the end, it *fails*
Any help is highly needed

aviallon, 2015-04-10:
```
echo "Test"
```

bits3rpent, 2015-04-12:
[jiv@Vulcan rtl8812au]$ cat /etc/os-release
NAME=Fedora
VERSION="21 (Twenty One)"
ID=fedora
VERSION_ID=21
PRETTY_NAME="Fedora 21 (Twenty One)"
ANSI_COLOR="0;34"
CPE_NAME="cpe:/o:fedoraproject:fedora:21"
HOME_URL="https://fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=21
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=21
[jiv@Vulcan rtl8812au]$ cat /etc/redhat-release
Fedora release 21 (Twenty One)

Works for me: last few lines
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/jiv/Downloads/rtl8812au/8812au.mod.o
  LD [M]  /home/jiv/Downloads/rtl8812au/8812au.ko
make[1]: Leaving directory '/usr/src/kernels/3.19.3-200.fc21.x86_64'

[jiv@Vulcan rtl8812au]$ yum info kernel-devel
Loaded plugins: langpacks
Installed Packages
Name        : kernel-devel
Arch        : x86_64
Version     : 3.17.4
Release     : 301.fc21
Size        : 34 M
Repo        : installed
From repo   : fedora
Summary     : Development package for building kernel modules to match the
            : kernel
URL         : http://www.kernel.org/
License     : GPLv2 and Redistributable, no modification permitted
Description : This package provides kernel headers and makefiles sufficient to
            : build modules against the kernel package.

Name        : kernel-devel
Arch        : x86_64
Version     : 3.19.3
Release     : 200.fc21
Size        : 35 M
Repo        : installed
From repo   : updates
Summary     : Development package for building kernel modules to match the
            : kernel
URL         : http://www.kernel.org/
License     : GPLv2 and Redistributable, no modification permitted
Description : This package provides kernel headers and makefiles sufficient to
            : build modules against the kernel package.
[jiv@Vulcan rtl8812au]$ yum info kernel-headers
Loaded plugins: langpacks
Installed Packages
Name        : kernel-headers
Arch        : x86_64
Version     : 3.19.3
Release     : 200.fc21
Size        : 3.3 M
Repo        : installed
From repo   : updates
Summary     : Header files for the Linux kernel for use by glibc
URL         : http://www.kernel.org/
License     : GPLv2 and Redistributable, no modification permitted
Description : Kernel-headers includes the C header files that specify the
            : interface between the Linux kernel and userspace libraries and
            : programs.  The header files define structures and constants that
            : are needed for building most standard programs and are also needed
            : for rebuilding the glibc package.

bits3rpent, 2015-04-12:
Please make sure kernel headers and kernel-devel is installed.

aviallon, 2015-04-16:
I can see x86_64 architecture, or my architecture is i686. It's a little bit old, but still supported. And yes kernel-headers and kernel-devel are installed on my computer.

aviallon, 2015-04-16:
But it works very well on my 64 bits PC

closed 2018-02-23 by aviallon