gnab/rtl8812au #41: 5GHz networks do not connect

SilverWingedSeraph, 2015-05-04:
I cannot connect to 5GHz. I'm on Ubuntu 15.04, kernel 3.19.0-15-generic. This is what happens when I try to connect:

    >tail -f /var/log/kern.log
    May  1 11:50:45 MinasTirith kernel: [ 2533.066569] RTL871X: ###=> usb_read_port_complete => urb status(-71)
    May  1 11:50:45 MinasTirith kernel: [ 2533.066953] RTL871X: ###=> usb_read_port_complete => urb status(-71)
    May  1 11:50:45 MinasTirith kernel: [ 2533.067331] RTL871X: ###=> usb_read_port_complete => urb status(-71)
    May  1 11:50:45 MinasTirith kernel: [ 2533.068620] RTL871X: ###=> usb_read_port_complete => urb status(-71)
    May  1 11:50:45 MinasTirith kernel: [ 2533.068963] RTL871X: ###=> usb_read_port_complete => urb status(-71)
    May  1 11:50:45 MinasTirith kernel: [ 2533.068964] RTL871X: [dvobj:ffff8803c41e6400][ERROR] continual_urb_error:5 > 4
    May  1 11:50:45 MinasTirith kernel: [ 2533.185869] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(0) OR bSurpriseRemoved(1) bReadPortCancel(0)
    May  1 11:50:45 MinasTirith kernel: [ 2533.288378] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(0) OR bSurpriseRemoved(1) bReadPortCancel(0)
    May  1 11:50:45 MinasTirith kernel: [ 2533.288464] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(0) OR bSurpriseRemoved(1) bReadPortCancel(0)
    May  1 11:50:46 MinasTirith kernel: [ 2534.729260] RTL871X: rtw_cmd_thread: DriverStopped(0) SurpriseRemoved(1) break at line 446

The driver crashes and I have to unplug and replug the adapter, an Alfa Networks AWUS036AC.

Chlorus, 2015-05-05:
Have a similar issue, with Linux Mint 17.1 Linux Kernel 3.13. Connection runs fine for roughly a minute or two then starts dropping all traffic. Works fine on 2.4GHz, albeit slowly.

SilverWingedSeraph, 2015-05-05:
I think your issue is somewhat different - can you post your
/var/log/kernel.log from around the time the connection drops?

Chlorus, 2015-05-08:
Upon looking at my log, it's actually a different issue. Please disregard my comment.