gnab/rtl8812au #48: Raspberry pi 2 support?

HaniKali, 2015-06-12:
Hey I tried using the make install method but i get the followoing error:

root@kali:~/rtl8812au-master# make make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/3.18.9-v7/build M=/root/rtl8812au-master modules make: * /lib/modules/3.18.9-v7/build: No such file or directory. Stop. make: * [modules] Error 2

Please can you help me sort it out or point me in the right direction thanks.

meoow, 2015-06-17:
You could try my repository, it was tested on Raspberry Pi 2 ArchLinux ARM.
[dkms-rtl8812au](https://github.com/meoow/dkms-rtl8812au)

meoow, 2015-06-17:
@HaniKali I see you kernel's version is 3.18.9-v7, so I'm assuming you are using Raspbian with stock kernel. Note that the stock kernel lacks certain modules for 8812au driver loading properly. You may want to build your own kernel with required modules (see the readme in my repository) or switching to another distro (like ArchLinux ARM)

HaniKali, 2015-06-17:
Hey thanks for your reply, I think I will just get a compatible card like the one from AWUS I had. Im not sure how I'd go with building my own kernel as I have zero experience.

yacob841, 2015-08-27:
I noticed this is way too late but for future people trying to solve this, it is because /lib/modules/$(uname -r)/build is a broken link.  So you need to delete it and ln to the kernel again.

Charlyo, 2017-04-14:
@yacob841  I'm facing the same issue. However, the link from my xbian /lib/modules/4.9.17+ isn't broken (can ls inside).