gnab/rtl8812au #49: Suggest turning off the debugging log by default

meoow, 2015-06-17:
Otherwise there will be massive amount of logs of the driver put into **dmesg**, it impacts the performance quite a lot.
Here is the patch:
```patch
--- autoconf.h	2015-06-17 16:01:11.000000000 +0800
+++ autoconf.h.patched	2015-06-17 16:00:40.000000000 +0800
@@ -378,14 +378,14 @@
 /*
  * Debug Related Config
  */
-#define DBG	1
+#define DBG	0
 
-#define CONFIG_DEBUG /* DBG_871X, etc... */
+//#define CONFIG_DEBUG /* DBG_871X, etc... */
 //#define CONFIG_DEBUG_RTL871X /* RT_TRACE, RT_PRINT_DATA, _func_enter_, _func_exit_ */
 
-#define CONFIG_PROC_DEBUG
+//#define CONFIG_PROC_DEBUG
 
-#define DBG_CONFIG_ERROR_DETECT
+//#define DBG_CONFIG_ERROR_DETECT
 //#define DBG_CONFIG_ERROR_DETECT_INT
 //#define DBG_CONFIG_ERROR_RESET
 ```

bit-pirate, 2015-08-07:
+1

There is so much output on dmesg that it makes it hard to monitor other parts of the system.

Suggest to add a compile flag to the Makefile to easily turn debugging on/off.

@meoow could you create a PR with your patch?

gnab, 2015-08-07:
Applied the patch. Please supply PR with compile flag if wanted.

fkpwolf, 2016-11-21:
There are still lots of log like:
```
[ 4771.233608] RTL871X: nolinked power save leave
[ 4779.301044] RTL871X: nolinked power save enter
```
To close this log:
```diff
diff --git a/core/rtw_pwrctrl.c b/core/rtw_pwrctrl.c
index ab995d6..89ebf85 100644
--- a/core/rtw_pwrctrl.c
+++ b/core/rtw_pwrctrl.c
@@ -46,7 +46,7 @@ void ips_enter(_adapter * padapter)
        if(rf_off == pwrpriv->change_rfpwrstate )
        {       
                pwrpriv->bpower_saving = _TRUE;
-               DBG_871X_LEVEL(_drv_always_, "nolinked power save enter\n");
+               DBG_871X_LEVEL(_drv_debug_, "nolinked power save enter\n");
 
                if(pwrpriv->ips_mode == IPS_LEVEL_2)
                        pwrpriv->bkeepfwalive = _TRUE;
@@ -81,7 +81,7 @@ int ips_leave(_adapter * padapter)
                if ((result = rtw_ips_pwr_up(padapter)) == _SUCCESS) {
                        pwrpriv->rf_pwrstate = rf_on;
                }
-               DBG_871X_LEVEL(_drv_always_, "nolinked power save leave\n");
+               DBG_871X_LEVEL(_drv_debug_, "nolinked power save leave\n");
 
                if((_WEP40_ == psecuritypriv->dot11PrivacyAlgrthm) ||(_WEP104_ == psecuritypriv->dot11PrivacyAlgrthm))
                {

```

ThomasCr, 2019-06-10:
is there a way, to tun debugging off on runtime or module loading?