gnab/rtl8812au #50: This driver looks like doesn't support "monitor" mode, does any one known about this ?

yangjunfei123, 2015-06-30:
I tried to set wifi mode to "Monitor", but it prompts "Invalid argument";  if I set mode to "Ad-Hoc", then it worked.  I jus want to know the chipset rtl8812 isn't support this feature from hardware, or this driver isn't support it.

The following is error message.
[root@localhost Downloads]# iwconfig enp0s26u1u6 mode Monitor
Error for wireless request "Set Mode" (8B06) :
    SET failed on device enp0s26u1u6 ; Invalid argument.

AaronMcHale, 2015-07-22:
It is supported but the rtl8812au driver doesn't communicate well with tools like iw, as they use nl80211, take a look at this repo https://github.com/jekader/hostapd-rtl, it adds support for this type of RealTek driver to hostapd which I'm assuming you are ultimately looking to use.

yangjunfei123, 2015-10-31:
Hi AaronMcHale
when I run "airmon start enp0s29u1u8 " I got this error,  got any idea with this ?

[root@localhost neil.yang]# airmon-ng start enp0s29u1u8
Found 5 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to kill (some of) them!

  PID Name
  683 avahi-daemon
  715 avahi-daemon
  811 NetworkManager
 1187 wpa_supplicant
 1584 dhclient

PHY	Interface	Driver		Chipset

null	enp0s29u1u8	rtl8812au	 
[root@localhost neil.yang]# airodump-ng --ivs enp0s29u1u8
ioctl(SIOCSIWMODE) failed: Invalid argument

ARP linktype is set to 1 (Ethernet) - expected ARPHRD_IEEE80211,
ARPHRD_IEEE80211_FULL or ARPHRD_IEEE80211_PRISM instead.  Make
sure RFMON is enabled: run 'airmon-ng start enp0s29u1u8 <#>'
Sysfs injection support was not found either.

[root@localhost neil.yang]#

nikicat, 2015-11-11:
Monitor mode is supported in [4.3.22](https://github.com/Grawp/rtl8812au_rtl8821au/tree/4.3.22) driver.

curtiszimmerman, 2016-06-18:
If you're looking for monitor mode support that will compile on Ubuntu/LM, [this version](https://github.com/Fredy72/rtl8821au) will get you what you want without updating kernel or gcc as required by other drivers that support monitor mode.

amadeuspzs, 2016-08-15:
I've managed to get monitor mode working for an ALFA AC1200 running RTL8812au using https://github.com/Grawp/rtl8812au_rtl8821au on Kali Linux.

keralo, 2017-02-17:
tell us how please

amadeuspzs, 2017-02-17:
```bash
git clone git@github.com:Grawp/rtl8812au_rtl8821au.git
cd rtl8812au_rtl8821au
make && sudo make install
sudo reboot
```

Florianwall, 2017-11-08:
hey amadeuspzs, 
when i tried to install the driver it always gives me some errors.. I am using Kali linux on a virtual machine and i tried to install several drivers but most of them seem to get me errors. The first one I installed worked but this version doesn't support the monitor mode ..
Can you tell me what i am doing wrong?
Getting these errors:

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-kali1-amd64/build M=/root/rtl8812au_rtl8821au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-kali1-amd64'
  CC [M]  /root/rtl8812au_rtl8821au/core/rtw_cmd.o
In file included from /root/rtl8812au_rtl8821au/include/drv_types.h:32:0,
                 from /root/rtl8812au_rtl8821au/core/rtw_cmd.c:22:
/root/rtl8812au_rtl8821au/include/osdep_service.h: In function ‘thread_enter’:
/root/rtl8812au_rtl8821au/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
/root/rtl8812au_rtl8821au/include/osdep_service.h: In function ‘flush_signals_thread’:
/root/rtl8812au_rtl8821au/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
/root/rtl8812au_rtl8821au/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.13.0-kali1-common/scripts/Makefile.build:307: recipe for target '/root/rtl8812au_rtl8821au/core/rtw_cmd.o' failed
make[4]: *** [/root/rtl8812au_rtl8821au/core/rtw_cmd.o] Error 1
/usr/src/linux-headers-4.13.0-kali1-common/Makefile:1529: recipe for target '_module_/root/rtl8812au_rtl8821au' failed
make[3]: *** [_module_/root/rtl8812au_rtl8821au] Error 2
Makefile:145: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-kali1-amd64'
Makefile:1670: recipe for target 'modules' failed
make: *** [modules] Error 2

rikichiman, 2017-11-25:
i have the same problem as  ( Florianwall )    :/ 
exactly the same errors..

kimocoder, 2017-11-25:
https://github.com/aircrack-ng/rtl8812au supports monitor mode + injection

curtiszimmerman, 2017-11-26:
Very nice! Thanks @kimocoder for the PR against the official repo and @astsam for the monitor mode support!