gnab/rtl8812au #54: Unable to compile on RPi2 with Arch Linux ARM after full upgrade

maziarzamani, 2015-08-02:
I made an full upgrade of my system and now i am unable to re-compile the module for rtl8812a.

Here is the full output from the make:

```
[root@test rtl8812au]# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.0.9-2-ARCH/build M=/opt/rtl8812au  modules
make[1]: Entering directory '/usr/lib/modules/4.0.9-2-ARCH/build'
  CC [M]  /opt/rtl8812au/os_dep/linux/rtw_android.o
/opt/rtl8812au/os_dep/linux/rtw_android.c: In function 'rtw_android_cmdstr_to_num':
/opt/rtl8812au/os_dep/linux/rtw_android.c:227:11: error: implicit declaration of function 'strnicmp' [-Werror=implicit-function-declaration]
   if(0 == strnicmp(cmdstr , android_wifi_cmd_str[cmd_num], strlen(android_wifi_cmd_str[cmd_num])) )
           ^
cc1: some warnings being treated as errors
scripts/Makefile.build:258: recipe for target '/opt/rtl8812au/os_dep/linux/rtw_android.o' failed
make[2]: *** [/opt/rtl8812au/os_dep/linux/rtw_android.o] Error 1
Makefile:1390: recipe for target '_module_/opt/rtl8812au' failed
make[1]: *** [_module_/opt/rtl8812au] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.0.9-2-ARCH/build'
Makefile:1051: recipe for target 'modules' failed
make: *** [modules] Error 2
```

donahue95, 2015-08-03:
https://github.com/csssuf/rtl8812au/archive/master.zip includes fixes for kernel-4; suggest using it doing whatever is needed in the makefile to configure it for raspberry pi.
https://github.com/gnab/rtl8812au is apparently inactive

maziarzamani, 2015-08-03:
That did the trick! Thanks.

closed 2015-08-03 by maziarzamani