gnab/rtl8812au #60: Possible error with RLT8821: Add Edimax EW-7811UAC discovery.

AussieInfoFinder, 2015-09-13:
Hi,

Perhaps it is necessary to add discovery of Edimax EW-7811UAC to 8812 and not 8821?

I have an Edimax EW-7811UAC, and I am typing this using this device as the wifi adapter, so - success! But, I needed to modify usb_intf.c.
Now, I'm not a developer - and hence haven't much clue what I'm doing, but - when I compiled the driver using the original usb_intf.c and installed the module and updated dependencies, no light flashed on the device and Mint 17 network management showed wireless with a cross. iwconfig showed everything off.

So, I added:
 {USB_DEVICE(0x7392, 0xA813),.driver_info = RTL8812}, /* Edimax - EW-7811UAC */
to the #ifdef CONFIG_RTL8812A section, and - hey presto - it worked!

I did this because lsusb returns:
Bus 002 Device 011: ID 7392:a812 Edimax Technology Co., Ltd 

Therefore, perhaps it is necessary to add discovery of this device to 8812 as well?

Sorry for any technical misunderstandings, J

fenixdoido, 2015-11-06:
You said to add this:

> {USB_DEVICE(0x7392, 0xA813),.driver_info = RTL8812}, /* Edimax - EW-7811UAC */

Because the lsusb return 7392: **a812**
So it should be:

> {USB_DEVICE(0x7392, **0xA812**),.driver_info = RTL8812}, /* Edimax - EW-7811UAC */

No? idk what im doing here :D

AussieInfoFinder, 2015-11-07:
Hi,

so first off, please confirm you have an Edimax  EW-7811UAC wifi adapter.
Have you downloaded the correct master.zip from github?
Have you tried to run 'make' to compile the driver?
If you have the EW-7811UAC and are completely in the dark regarding how to compile the driver yourself I could send you a link to my driver so you can download it.
If you want to do this, do you know how to install the driver once you have it?

let me know where you're at with this, J

fenixdoido, 2015-11-08:
Yes, I have the EW-7811UAC.
Mine works with gnab drivers, but, the led doesnt works (Any fix?)
Also, i cant go into monitor mode. Its a know problem right?

AussieInfoFinder, 2015-11-09:
Well, my led works. I compiled the driver on the same machine I use the Edimax with, I'm running Mint 17.2, which uses the 3.16.0-38-generic kernel.

If you're using the same kernel I could make my driver available for download so you could try it.
I'm pasting below the text from 'how to install' in case you haven't seen this yet. There may be something in it which is different to what you are doing to install the driver.

-------------------------
OK,

zip file with make script comes from Github, it's the one that is a fork for compiling drivers on kernel 3.0 and above for Realtek 8812au

copy the 8812au.ko version compiled on the Mint 17.2 (kernel 3.16) machine to: 
/lib/modules/3.16.0-38-generic/kernel/drivers/net/wireless/

sudo insmod 8812au.ko
ls mod | grep 8812au
if there, then
depmod (updates module dependencies so always loads at startup)
plug in the Edimax, enjoy!

kernel version : uname -r
lsusb shows the Edimax is an 8812 device
modinfo shows the devices supported
------------------
Let me know if you want to try my driver, J