gnab/rtl8812au #68: Auth succeeds, but assosiation fails

tosiara, 2015-12-19:
I read there had been issues running rtl8812au on ARM devices. Maybe this is one of them

Using [ODROID C1+](http://www.hardkernel.com/main/products/prdt_info.php?g_code=G143703355573&tab_idx=2) with kernel:
```
$ uname -a
Linux odroidc1p 3.10.80-135 #1 SMP PREEMPT Sat Nov 28 13:10:52 BRST 2015 armv7l armv7l armv7l GNU/Linux
```
Built fine. 
When starting wpa_supplicant it forver tries to associate. There is this sequence in the syslog:
```
[ 2608.389953@3] RTL871X: set bssid:60:e3:27:b8:37:60
[ 2608.410132@3] RTL871X: start auth
[ 2608.412595@3] RTL871X: auth success, start assoc
[ 2608.415314@3] RTL871X: assoc success
[ 2608.415852@0] RTL871X: indicate disassoc
[ 2608.417028@0] RTL871X: set bssid:00:00:00:00:00:00
```
The same dongle and same code tree works fine on x86_64 machine with kernel 4.1.13

Please let me know what kind of debugging I may perform on my ARM device to contribute
Thanks

tosiara, 2015-12-19:
Successful association on x86_64 machine has these corresponding syslog lines:
```
[114378.535292] RTL871X: set bssid:60:e3:27:b8:37:60
[114378.710724] RTL871X: start auth
[114378.713780] RTL871X: auth success, start assoc
[114378.716533] RTL871X: assoc success
[114378.717041] UpdateHalRAMask8812A => mac_id:0, networkType:0x14, mask:0x000ffff0
         ==> rssi_level:0, rate_bitmap:0x000ff000
[114378.720374] RTL871X: send eapol packet
[114378.723672] RTL871X: send eapol packet
[114378.724245] RTL871X: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) camid:4
[114378.724249] RTL871X: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:2
[114379.735446] UpdateHalRAMask8812A => mac_id:0, networkType:0x14, mask:0x000ffff0
         ==> rssi_level:1, rate_bitmap:0x000f0000
```

Half-Shot, 2015-12-24:
I've got a similiar issue with my x86_64 machine:

```
[   47.049705] RTL871X: nolinked power save leave
[   47.049729] RTL871X: set bssid:00:00:00:00:00:00
[   47.049751] RTL871X: set ssid [g\xffffffc6isQ\xffffffffJ\xffffffec)ͺ\xffffffab\xfffffff2\xfffffffb\xffffffe3F|\xffffffc2T\xfffffff8\xffffffe8\xffffffe7\xffffff8dvZ.c3\xffffff9fɚ\xffffff91\xfffffffb\xffffffeeq] fw_state=0x00000008
[   50.477688] RTL871X: set bssid:00:00:00:00:00:00
\xffffffb71X\xffffffa3Z%]X\xffffffe9^ԫ\xffffffb2\xffffffcdƛ\xffffffb4T\xffffff82tA!=܇\xffffffe1\xffffffe0\xffffffa2] fw_state=0x00000808
[   50.495730] RTL871X: indicate disassoc
[   55.065295] RTL871X: nolinked power save enter
[   60.517184] IPv6: ADDRCONF(NETDEV_UP): enp3s0u1: link is not ready
[   60.989990] RTL871X: nolinked power save leave
[   60.990010] RTL871X: set bssid:00:00:00:00:00:00
[   60.990028] RTL871X: set ssid [g\xffffffc6isQ\xffffffffJ\xffffffec)ͺ\xffffffab\xfffffff2\xfffffffb\xffffffe3F|\xffffffc2T\xfffffff8\xffffffe8\xffffffe7\xffffff8dvZ.c3\xffffff9fɚ1\xfffffffb\xffffffe9] fw_state=0x00000008
[   64.464804] RTL871X: set ssid [Sophie] fw_state=0x00000808
[   64.464817] RTL871X: set bssid:04:a1:51:df:e0:4b
[   65.418613] RTL871X: start auth
[   65.420715] RTL871X: auth success, start assoc
[   65.422687] RTL871X: assoc success
[   65.423922] UpdateHalRAMask8812A => mac_id:0, networkType:0x44, mask:0xfffffff0
               	 ==> rssi_level:0, rate_bitmap:0xfffff010
[   65.452317] IPv6: ADDRCONF(NETDEV_CHANGE): enp3s0u1: link becomes ready
[   65.453741] RTL871X: send eapol packet
[   65.992861] UpdateHalRAMask8812A => mac_id:0, networkType:0x44, mask:0xfffffff0
               	 ==> rssi_level:2, rate_bitmap:0xfffff000
[   66.451515] RTL871X: send eapol packet
[   67.449514] RTL871X: send eapol packet
[   68.447675] RTL871X: send eapol packet
[   69.445730] RTL871X: send eapol packet
[   70.443898] RTL871X: send eapol packet
[   71.442064] RTL871X: send eapol packet
[   72.439953] RTL871X: send eapol packet
[   73.437151] RTL871X: sta recv deauth reason code(15) sta:04:a1:51:df:e0:4b
[   73.447894] RTL871X: indicate disassoc
[   79.019607] RTL871X: nolinked power save enter
[   80.545203] RTL871X: nolinked power save leave
[   80.545207] RTL871X: set bssid:00:00:00:00:00:00
\xffffffb71X\xffffffa3Z%]X\xffffffe9^ԫ\xffffffb2\xffffffcdƛ\xffffffb4T\xffffff82tA!=܇b\xffffff83\xffffff91] fw_state=0x00000008
[   80.556751] RTL871X: indicate disassoc
[   86.551932] RTL871X: nolinked power save enter
[  109.612789] IPv6: ADDRCONF(NETDEV_UP): enp3s0u1: link is not ready
```

paralin, 2016-06-08:
? Was there ever a solution on this?