gnab/rtl8812au #76: Fails to compile on raspberry pi 2 "Raspbian GNU/Linux 8 (jessie)"

skandragon, 2016-03-11:
Linux raspberrypi 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux

    root@raspberrypi:/home/pi/rtl8812au# make
    make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.1.18-v7+/build M=/home/pi/rtl8812au  modules
    make[1]: Entering directory '/lib/modules/4.1.18-v7+/build'
    make[1]: *** No rule to make target 'modules'.  Stop.
    make[1]: Leaving directory '/lib/modules/4.1.18-v7+/build'
    Makefile:1052: recipe for target 'modules' failed
    make: *** [modules] Error 2

vassilmladenov, 2017-02-14:
Did you ever figure out this issue?

Charlyo, 2017-04-14:
Having the same issue with xbian and kernel 4.9.17

CRTX, 2018-01-10:
This is the same issue as this: https://github.com/gnab/rtl8812au/issues/129

Fixed by `sudo apt-get install raspberrypi-kernel-headers`