gnab/rtl8812au #80: No device after upgrading kernel to 4.4.6-301.fc23.x86_64

abc-mikey, 2016-04-09:
Hi, 

Having issues here after an upgrade to kernel 4.4.6-301.fc23.x86_64.

The module was rebuilt and appears to load but there is no apparent activity from the device. 

```
grep 8812 out-lsmod.txt 
8812au                991232  0
```

dmesg is giving the following during a reboot:

```
grep -A 30 '\(rtl\|8812\)' out-dmesg.txt 
[    3.204580] usb 1-1: New USB device found, idVendor=0bda, idProduct=8812
[    3.204581] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.204582] usb 1-1: Product: 802.11n NIC
[    3.204584] usb 1-1: Manufacturer: Realtek
[    3.204585] usb 1-1: SerialNumber: 123456
[    3.204609] hub 2-4:1.0: 7 ports detected
[    3.228279] ata3: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[    3.228306] ata4: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[    3.228641] ata3.00: ATA-8: CSSD-V60GB2, 1.0, max UDMA/100
[    3.228643] ata3.00: 117231408 sectors, multi 16: LBA48 NCQ (depth 31/32), AA
[    3.229224] ata3.00: configured for UDMA/100
[    3.229266] ata2: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[    3.229947] ata4.00: supports DRM functions and may not be fully accessible
[    3.230084] ata4.00: READ LOG DMA EXT failed, trying unqueued
[    3.230133] ata4.00: failed to get NCQ Send/Recv Log Emask 0x1
[    3.230135] ata4.00: ATA-9: Samsung SSD 850 EVO 250GB, EMT01B6Q, max UDMA/133
[    3.230136] ata4.00: 488397168 sectors, multi 1: LBA48 NCQ (depth 31/32), AA
[    3.230612] ata4.00: supports DRM functions and may not be fully accessible
[    3.230699] ata4.00: failed to get NCQ Send/Recv Log Emask 0x1
[    3.230806] ata4.00: configured for UDMA/133
[    3.232641] ata2.00: ATA-7: SAMSUNG SP2504C, VT100-52, max UDMA7
[    3.232643] ata2.00: 488281250 sectors, multi 16: LBA48 NCQ (depth 31/32), AA
[    3.248393] ata2.00: configured for UDMA/133
[    3.248591] scsi 1:0:0:0: Direct-Access     ATA      SAMSUNG SP2504C  0-52 PQ: 0 ANSI: 5
[    3.248808] sd 1:0:0:0: [sda] 488281250 512-byte logical blocks: (250 GB/233 GiB)
[    3.248829] sd 1:0:0:0: Attached scsi generic sg0 type 0
[    3.248909] sd 1:0:0:0: [sda] Write Protect is off
[    3.248911] sd 1:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    3.248950] sd 1:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    3.249008] scsi 2:0:0:0: Direct-Access     ATA      CSSD-V60GB2      1.0  PQ: 0 ANSI: 5
[    3.249139] sd 2:0:0:0: Attached scsi generic sg1 type 0
--
[    7.701553] usbcore: registered new interface driver rtl8812au
[    7.708257] rtl8812au 1-1:1.0 enp0s18f2u1: renamed from wlan0
[    8.486812] nf_conntrack version 0.5.0 (16384 buckets, 65536 max)
[    8.514656] ip6_tables: (C) 2000-2006 Netfilter Core Team
[    8.615205] Ebtables v2.0 registered
[    8.643579] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[    8.979844] Netfilter messages via NETLINK v0.30.
[    8.994669] ip_set: protocol 6
[    9.201652] IPv6: ADDRCONF(NETDEV_UP): enp2s0: link is not ready
[    9.344991] r8169 0000:02:00.0 enp2s0: link down
[    9.345060] IPv6: ADDRCONF(NETDEV_UP): enp2s0: link is not ready
[    9.345112] cfg80211: World regulatory domain updated:
[    9.345122] cfg80211:  DFS Master region: unset
[    9.345127] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    9.345136] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    9.345143] cfg80211:   (2457000 KHz - 2482000 KHz @ 20000 KHz, 92000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    9.345149] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[    9.345155] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    9.345161] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    9.345166] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[    9.345171] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[    9.345176] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   10.102297] tun: Universal TUN/TAP device driver, 1.6
[... 68 lines cut ...]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0): Activation: starting connection 'virbr0' (d2134bd4-ad0b-45c1-9724-499b1a81891b)
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0): device state change: disconnected -> prepare (reason 'none') [30 40 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0): device state change: prepare -> config (reason 'none') [40 50 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  keyfile: add connection in-memory (42f1be07-edac-4d38-ba56-91d987b2f206,"virbr0-nic")
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): device state change: unmanaged -> unavailable (reason 'connection-assumed') [10 20 41]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): device state change: unavailable -> disconnected (reason 'connection-assumed') [20 30 41]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): Activation: starting connection 'virbr0-nic' (42f1be07-edac-4d38-ba56-91d987b2f206)
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0): device state change: config -> ip-config (reason 'none') [50 70 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): device state change: disconnected -> prepare (reason 'none') [30 40 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0): device state change: ip-config -> ip-check (reason 'none') [70 80 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): device state change: prepare -> config (reason 'none') [40 50 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): device state change: config -> ip-config (reason 'none') [50 70 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): device state change: ip-config -> secondaries (reason 'none') [70 90 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): device state change: secondaries -> activated (reason 'none') [90 100 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  NetworkManager state is now CONNECTED_LOCAL
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): Activation: successful, device activated.
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  connectivity: check for uri 'http://fedoraproject.org/static/hotspot.txt' failed with 'Error resolving 'fedoraproject.org': No address associated with hostname'
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0-nic): link disconnected (deferring action for 4 seconds)
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0): device state change: ip-check -> secondaries (reason 'none') [80 90 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0): device state change: secondaries -> activated (reason 'none') [90 100 0]
Apr 09 11:40:51 nixon NetworkManager[1041]: <info>  (virbr0): Activation: successful, device activated.
Apr 09 11:40:56 nixon NetworkManager[1041]: <info>  startup complete
Apr 09 11:40:56 nixon NetworkManager[1041]: <info>  (virbr0-nic): link disconnected (calling deferred action)
Apr 09 11:45:52 nixon NetworkManager[1041]: <info>  connectivity: check for uri 'http://fedoraproject.org/static/hotspot.txt' failed with 'Error resolving 'fedoraproject.org': No address associated with hostname'
```

abc-mikey, 2016-04-09:
This line seems suspicious: 
```
[    7.708257] rtl8812au 1-1:1.0 enp0s18f2u1: renamed from wlan0
```
Why does it not start with 'wl'? 

If I check `ifconfig` I can see it:

```
ifconfig -a
enp0s18f2u1: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 04:8d:38:d6:b7:a4  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

enp2s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 1c:6f:65:8a:c8:8e  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 7021  bytes 755915 (738.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7021  bytes 755915 (738.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

virbr0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.122.1  netmask 255.255.255.0  broadcast 192.168.122.255
        ether 52:54:00:64:60:19  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

virbr0-nic: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 52:54:00:64:60:19  txqueuelen 500  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp0s18f2u2u4: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.10  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::287:32ff:fe88:552  prefixlen 64  scopeid 0x20<link>
        ether 00:87:32:88:05:52  txqueuelen 1000  (Ethernet)
        RX packets 336566  bytes 59514513 (56.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 463169  bytes 74648513 (71.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

abc-mikey, 2016-04-09:
This line seems suspicious: 
```
[    7.708257] rtl8812au 1-1:1.0 enp0s18f2u1: renamed from wlan0
```
Why does it not start with 'wl', it should be detected as a wifi device, right? 

If I check `ifconfig` I can see it:

```
ifconfig -a
enp0s18f2u1: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 04:8d:38:d6:b7:a4  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

enp2s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 1c:6f:65:8a:c8:8e  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 7021  bytes 755915 (738.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7021  bytes 755915 (738.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

virbr0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.122.1  netmask 255.255.255.0  broadcast 192.168.122.255
        ether 52:54:00:64:60:19  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

virbr0-nic: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 52:54:00:64:60:19  txqueuelen 500  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp0s18f2u2u4: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.10  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::287:32ff:fe88:552  prefixlen 64  scopeid 0x20<link>
        ether 00:87:32:88:05:52  txqueuelen 1000  (Ethernet)
        RX packets 336566  bytes 59514513 (56.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 463169  bytes 74648513 (71.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

abc-mikey, 2016-04-09:
Rebooting with `net.ifnames=0` does stop it being renamed but it is still not listed as active on `ifconfig` although it now shows on `ifconfig -a` as `wlan 0`.

```
sudo ifconfig -a
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 1c:6f:65:8a:c8:8e  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 1359  bytes 173793 (169.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1359  bytes 173793 (169.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

virbr0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.122.1  netmask 255.255.255.0  broadcast 192.168.122.255
        ether 52:54:00:64:60:19  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

virbr0-nic: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 52:54:00:64:60:19  txqueuelen 500  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 04:8d:38:d6:b7:a4  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

abc-mikey, 2016-04-09:
And I get the following from `dmesg -c` after running `sudo ifconfig wlan0 up`.

```
[  268.963821] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  270.968144] RTL871X: nolinked power save enter
```

abc-mikey, 2016-04-09:
`iw dev` and `rfkill list all` do not see it:

```
iw dev
rfkill list all
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
```

abc-mikey, 2016-04-09:
```
lsusb | grep 8812
Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
```

Just in case it wasn't clear from what I have already posted here is the device I have.

simonjm, 2016-04-14:
I ran into something similar. Downgrading NetworkManager to 1.0.10-2 fixed it for me. See #81

abc-mikey, 2016-04-14:
Hi @simonjm,

Yes I can confirm that running the following on my fedora 23 system reverses the issue:

```
sudo dnf downgrade --allowrasing NetworkManager NetworkManager-wwan NetworkManager-wifi NetworkManager-team NetworkManager-bluetooth NetworkManager-adsl  NetworkManager-libnm
```

Here are the 2 versions of NetworkManager, the one I've downgraded to and the one I was seeing the issue with:

```
dnf info NetworkManager
Last metadata expiration check: 0:09:51 ago on Thu Apr 14 06:35:03 2016.
Installed Packages
Name        : NetworkManager
Arch        : x86_64
Epoch       : 1
Version     : 1.0.6
Release     : 6.fc23
Size        : 9.2 M
Repo        : @System
From repo   : fedora
Summary     : Network connection manager and user applications
URL         : http://www.gnome.org/projects/NetworkManager/
License     : GPLv2+
Description : NetworkManager is a system service that manages network interfaces
            : and connections based on user or automatic configuration. It
            : supports Ethernet, Bridge, Bond, VLAN, Team, InfiniBand, Wi-Fi,
            : mobile broadband (WWAN), PPPoE and other devices, and supports a
            : variety of different VPN services.

Available Packages
Name        : NetworkManager
Arch        : x86_64
Epoch       : 1
Version     : 1.0.10
Release     : 3.fc23
Size        : 1.9 M
Repo        : updates
Summary     : Network connection manager and user applications
URL         : http://www.gnome.org/projects/NetworkManager/
License     : GPLv2+
Description : NetworkManager is a system service that manages network interfaces
            : and connections based on user or automatic configuration. It
            : supports Ethernet, Bridge, Bond, VLAN, Team, InfiniBand, Wi-Fi,
            : mobile broadband (WWAN), PPPoE and other devices, and supports a
            : variety of different VPN services.
```

abc-mikey, 2016-04-16:
Obviously a permanent downgrade is not a solution.

abc-mikey, 2016-04-22:
OK, I've switched to https://github.com/abperiasamy/rtl8812AU_8821AU_linux/graphs/contributors

closed 2016-04-22 by abc-mikey