gnab/rtl8812au #82: Truncated packets?

aarond10, 2016-04-17:
I suspect the rtl8812au driver is truncating some of my packets. I'm seeing this:

```
0000   f3 9f 81 80 00 01 00 02 00 00 00 00 03 77 77 77  .............www
0010   0a 74 65 6e 73 6f 72 66 6c 6f 77 03 6f 72 67 00  .tensorflow.org.
0020   00 01 00 01 c0 0c 00 05 00 01 00 00 00 4d 00 16  .............M..
0030   03 67 68 73 0c 67 6f 6f 67 6c 65 68 6f 73 74 65  .ghs.googlehoste
0040   64 03 63 6f 6d 00 c0 30 00 01 00 01 00 00 00 46  d.com..0.......F
0050   00 04                                            ..
```

Where I should be seeing this:

```
0000   f1 73 81 80 00 01 00 02 00 00 00 00 03 77 77 77  .s...........www
0010   0a 74 65 6e 73 6f 72 66 6c 6f 77 03 6f 72 67 00  .tensorflow.org.
0020   00 01 00 01 c0 0c 00 05 00 01 00 00 01 2b 00 16  .............+..
0030   03 67 68 73 0c 67 6f 6f 67 6c 65 68 6f 73 74 65  .ghs.googlehoste
0040   64 03 63 6f 6d 00 c0 30 00 01 00 01 00 00 01 2b  d.com..0.......+
0050   00 04 <b>40 e9 bc 79</b>                                ..<b>@..y</b>
```

This is a standard DNS request. Strangely, I get the right data sometimes, and in particular, if I use "dig" I get seem to get the right response, if I use chrome, I get a truncated response and "NXDOMAIN_ERROR".

Not limited to just this one domain either. I'm seeing it for youtube.com and other domains too. And the request coming out of chrome doesn't seem borked in any way.

Both the IP and UDP headers report the correct size (114 above), but the packet is truncated by 4 bytes "on the wire".

Running ath10k openwrt on the server side. 80Mhz channel 44, WPA2 PSK (TKIP,CCMP).
Problem goes away when I change to 2.4Ghz radio but doesn't present on my Macbook at 5Ghz, nor any of the 4 android devices I own so while I can't entirely rule out the ath10k side, I strongly suspect framing issues in the rtl8812au driver.

pcap file attached showing bad responses. Acquired with tcpdump -s 2000 -w /tmp/pcap.

[pcap.zip](https://github.com/gnab/rtl8812au/files/222583/pcap.zip)

aarond10, 2016-04-17:
Can confirm I don't see any truncation if I lock bandwidth to 40Mhz. Looking very likely that 80Mhz codepath bug somewhere.

aarond10, 2016-04-17:
Actual devices I'm using here are a TP-Link Archer C7 running OpenWRT head from about a month ago and a Dlink DWA182 - C1 (USB ID 2001:3315). Have flipped back and forth. Possibly only seeing this issue at 867Mbps it seems. Dropping to 40Mhz fixed it. Tried playing with /sys/modules/8812au/paramaters/* values but had no effect. Also tried looking at code but I don't know heads from tails without spending hours learning it first.

blueponies666, 2016-08-22:
Same problem, my SSH connections keep disconnecting, and TCP retransmissions often appear. WIRESHARK says something about an incorrect packet length.

stevedee, 2017-02-15:
I think my problem with BBC internet radio streams is related to this (OK on 2.4GHz, but drops after a few seconds on 5GHz). Reducing channel width to 40MHz does not fix in my case. Reducing RTS threshold to 500 helps but not enough to be a fix.

Problem occurs on Lubuntu Linux and Raspbian RPi.

More info: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=149242&p=981452#p981452

mayli, 2018-02-15:
@aarond10 how did you limit the bandwidth to 40Mhz? I saw same behavior, but i am not quite familiar with `iwconfig`.

aarond10, 2018-02-15:
Depends on the version of your software but try either this (the MCS stuff):
https://wireless.wiki.kernel.org/en/users/Documentation/iw#modifying_transmit_bitrates

or this (HT40, etc):
http://trac.gateworks.com/wiki/wireless/wifi