gnab/rtl8812au #86: make error with os_intfs.c

M1k3CrU2, 2016-06-02:
New to Linux, using centos 7 
When I make rtl8812au, I get these two errors at the end:

/rtl8812au/os_dep/linux/os_intfs.c:1002:92: error: unknown type name ‘select_queue_fallback_t’
 static u16 rtw_select_queue(struct net_device *dev, struct sk_buff *skb, void *accel_priv, select_queue_fallback_t fallback)
                                                                                            ^
/rtl8812au/os_dep/linux/os_intfs.c:1053:22: error: ‘rtw_select_queue’ undeclared here (not in a function)
  .ndo_select_queue = rtw_select_queue,
                      ^
make[2]: *** [/rtl8812au/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [_module_/rtl8812au] Error 2

myly10, 2016-07-24:
#91

myly10, 2016-07-24:
#91 
I've raised a pull request that can fix this issue.