gnab/rtl8812au #87: Unable to go to Monitor mode

syndrid, 2016-06-11:
When I type
`sudo iwconfig wlan1 mode monitor`
I get:
`Error for wireless request "Set Mode" (8B06) :+1: 
SET failed on device wlan ; Invalid argument.`

Other modes work:
`sudo iwconfig wlan1 mode managed`

curtiszimmerman, 2016-06-18:
I can't put my device in monitor mode, either.

rzwei, 2016-10-11:
Me either.

DeveloppSoft, 2016-12-27:
I am to.
If I look at the sources, the driver seems to support monitor mode:
```shell
eliott@tux:~/Documents/ClefWifi/rtl8812au$ grep "monitor" -R ./
./hal/rtl8812a/usb/usb_halinit.c: *			need to monitor the influence of FTP/network share.
./hal/OUTSRC/odm_debug.h:	u4Byte	Pm;			/* Performance monitor module. */	
./hal/OUTSRC/odm.c:				//3 If pre_state_flag=1 ==> start to monitor TP degrade 20%
./os_dep/linux/ioctl_linux.c:	{	//	0 is the default value and it means the application monitors the HW PBC doesn't privde its pid to driver.
./os_dep/linux/ioctl_cfg80211.c:static int rtw_cfg80211_monitor_if_open(struct net_device *ndev)
./os_dep/linux/ioctl_cfg80211.c:static int rtw_cfg80211_monitor_if_close(struct net_device *ndev)
./os_dep/linux/ioctl_cfg80211.c:static int rtw_cfg80211_monitor_if_xmit_entry(struct sk_buff *skb, struct net_device *ndev)
./os_dep/linux/ioctl_cfg80211.c:static void rtw_cfg80211_monitor_if_set_multicast_list(struct net_device *ndev)
./os_dep/linux/ioctl_cfg80211.c:static int rtw_cfg80211_monitor_if_set_mac_address(struct net_device *ndev, void *addr)
./os_dep/linux/ioctl_cfg80211.c:static const struct net_device_ops rtw_cfg80211_monitor_if_ops = {
./os_dep/linux/ioctl_cfg80211.c:	.ndo_open = rtw_cfg80211_monitor_if_open,
./os_dep/linux/ioctl_cfg80211.c:       .ndo_stop = rtw_cfg80211_monitor_if_close,
./os_dep/linux/ioctl_cfg80211.c:       .ndo_start_xmit = rtw_cfg80211_monitor_if_xmit_entry,
./os_dep/linux/ioctl_cfg80211.c:       .ndo_set_multicast_list = rtw_cfg80211_monitor_if_set_multicast_list,
./os_dep/linux/ioctl_cfg80211.c:       .ndo_set_mac_address = rtw_cfg80211_monitor_if_set_mac_address,       
./os_dep/linux/ioctl_cfg80211.c:static int rtw_cfg80211_add_monitor_if(_adapter *padapter, char *name, struct net_device **ndev)
./os_dep/linux/ioctl_cfg80211.c:		DBG_871X(FUNC_ADPT_FMT" monitor interface exist: "NDEV_FMT"\n",
./os_dep/linux/ioctl_cfg80211.c:	mon_ndev->netdev_ops = &rtw_cfg80211_monitor_if_ops;
./os_dep/linux/ioctl_cfg80211.c:	mon_ndev->open = rtw_cfg80211_monitor_if_open;
./os_dep/linux/ioctl_cfg80211.c:	mon_ndev->stop = rtw_cfg80211_monitor_if_close;
./os_dep/linux/ioctl_cfg80211.c:	mon_ndev->hard_start_xmit = rtw_cfg80211_monitor_if_xmit_entry;
./os_dep/linux/ioctl_cfg80211.c:	mon_ndev->set_mac_address = rtw_cfg80211_monitor_if_set_mac_address;
./os_dep/linux/ioctl_cfg80211.c:		ret = rtw_cfg80211_add_monitor_if(padapter, (char *)name, &ndev);
./os_dep/linux/ioctl_cfg80211.c:		DBG_871X(FUNC_NDEV_FMT" remove monitor interface\n", FUNC_NDEV_ARG(ndev));
./os_dep/linux/ioctl_cfg80211.c:		DBG_8192C("%s, unregister monitor interface\n", __func__);
./include/ioctl_cfg80211.h:	struct net_device *pmon_ndev;//for monitor interface
./include/ioctl_cfg80211.h:	char ifname_mon[IFNAMSIZ + 1]; //interface name for monitor interface
./include/rtw_cmd.h:#define IW_MODE_MONITOR	6	// Passive monitor (listen only)
./include/Hal8192CPhyCfg.h:// BB/MAC/RF other monitor API
./include/Hal8192EPhyCfg.h:// BB/MAC/RF other monitor API
./include/Hal8812PhyCfg.h:// BB/MAC/RF other monitor API
./include/Hal8192DPhyCfg.h:// BB/MAC/RF other monitor API
./include/Hal8188EPhyCfg.h:// BB/MAC/RF other monitor API
./core/rtw_mlme_ext.c:			/*to monitor whether the AP is alive or not*/
eliott@tux:~/Documents/ClefWifi/rtl8812au$ 
```
Seems like it uses cfg80211...
BTW, I have another question, if you look at the following `./include/rtw_cmd.h:#define IW_MODE_MONITOR	6	// Passive monitor (listen only)`, you cannot inject during monitor mode, why?

JayTurtle, 2017-06-07:
I am having the same issue, any progress on this yet?

raphaelmerx, 2018-12-24:
Had the same problem, resolved it by using the driver from https://github.com/astsam/rtl8812au instead. That package is also available as a package for Kali.

So:
- remove this driver: `dkms remove -m 8812au -v 4.2.2 --all`
- on Kali: `apt install realtek-rtl88xxau-dkms`

After a reboot, `airmon-ng start wlan0` successfully starts monitor mode.

kimocoder, 2018-12-24:
Just so you know.. Kali packs the driver from https://github.com/aircrack-ng/rtl8812au and the v5.2.20(.2) version is used, not the v5.1.5 anymore (deprecated)

DanteKanji, 2020-09-18:
> Had the same problem, resolved it by using the driver from https://github.com/astsam/rtl8812au instead. That package is also available as a package for Kali.
>
> So:
>
> * remove this driver: `dkms remove [...]

thank you.  this is the only thing that worked for me after trying to fix this issue for over 12 hours.