gnab/rtl8812au #90: Breaks for kernel version 4.6

spwhitton, 2016-07-17:
My device can no longer connect with this new kernel version.  It would be great if you could update the driver.  Thanks!

zebulon2, 2016-07-21:
Same problem here on an Arch 64 installation. Actually, the driver was working up to 4.6.3, but is broken with 4.6.4. It compiles, but it times out when trying to connect.

intdel, 2016-07-25:
I am using it with 4.7. Works just fine for me.

spwhitton, 2016-07-26:
Fails with 4.7.0-rc7 here.

intdel, 2016-07-26:
Do you have any error messages or something? Btw. 4.7 is out maybe they fixed a bug, so try it with that. Also make sure wifi is actually enabled in your kernel config.

spwhitton, 2016-07-26:
Good point.  When 4.7 reaches Debian experimental I'll try it out.

No errors in dmesg.  I just don't see an interface.

zebulon2, 2016-07-26:
Just tried again with success. I do not know what happened, kernel name is still 4.6.4 (-1 on Arch 64). Possibly a dkms issue.

spwhitton, 2016-11-21:
Now working on 4.8.  I think that the issue was with dkms, rather than this module.  Thanks for the feedback.

closed 2016-11-21 by spwhitton