gnab/rtl8812au #92: kernel 3.14.26 :  Unable to handle kernel NULL pointer dereference at virtual address 00000000

antismap, 2016-08-01:
Hello everybody,

It doesn't work for me on Openwrt chaos calmer, kernel  3.14.26
The kernel Oopses as soon as I plug in my USB stick. I tested two sticks with the same chip : edimax EW-7822UAC and TP-Link T4U
 
I compiled the driver and inserted it with insmod. Did I forget something ? If I understood everything, no firmware file is required ? 
I'm stuck on this since a few days, thank you for your help !

`   82.537949] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   82.546576] pgd = c0004000
[   82.549431] [00000000] *pgd=00000000
[   82.553209] Internal error: Oops: 817 [#1] ARM
[   82.557874] Modules linked in: 8812au option iptable_nat usb_wwan sierra pppoe nf_nat_ipv4 nf_conntrack_netlink nf_conntrack_ipv6 nf_conntrack_ipv4 mcs7830 ipt_MASQUERADE ftdi_sio cdc_ether xt_time xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda pptp pppox ppp_mppe ppp_async nfnetlink nf_nat_irc nf_nat_ftp nf_nat nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_irc nf_conntrack_ftp nf_conntrack iptable_raw iptable_mangle iptable_filter ipt_REJECT ip_tables fuse sch_teql sch_tbf sch_sfq sch_red sch_prio sch_htb sch_gred sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_police act_ipt act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress musb_am335x musb_dsps musb_hdrc udc_core phy_am335x phy_am335x_control phy_generic ip6t_REJECT ip6table_raw ip6table_mangle ip6table_filter ip6_tables swconfig pppoatm ppp_generic slhc nfsd ip_gre gre tun cifs dm_crypt dm_mirror dm_region_hash dm_log dm_mod br2684 atm nls_utf8 nls_cp850 cryptosoft cryptodev ocf md4 usb_storage ehci_platform ehci_hcd usbcore usb_common exportfs mii
[   82.678036] CPU: 0 PID: 768 Comm: khubd Not tainted 3.14.26 #17
[   82.684259] task: cd370100 ti: cd554000 task.ti: cd554000
[   82.689955] PC is at memcpy+0xb4/0x330
[   82.693941] LR is at 0x196df2f4
[   82.697251] pc : [<c000d1d4>]    lr : [<196df2f4>]    psr: 00010013
[   82.697251] sp : cd555c34  ip : 0000001c  fp : 00004164
[   82.709305] r10: 000040b4  r9 : 00000006  r8 : cc326e00
[   82.714794] r7 : cc326e00  r6 : 00000000  r5 : cc32a000  r4 : d12f8000
[   82.721651] r3 : 00000000  r2 : ffffffe6  r1 : d12fb1ec  r0 : 00000000
[   82.728511] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[   82.736188] Control: 10c5387d  Table: 8c438019  DAC: 00000015
[   82.742225] Process khubd (pid: 768, stack limit = 0xcd554238)
[   82.748352] Stack: (0xcd555c34 to 0xcd556000)
[   82.752939] 5c20:                                              cc32a000 00000000 cc326e00
[   82.761539] 5c40: cc326e00 00000000 d12f8000 bf37520c 00000000 bf3766f0 d12fb100 bf3b2a88
[   82.770140] 5c60: 000000f4 d12f8000 bf3b8fa8 cc26ca00 cc26ca20 00000002 00000035 cc1a9068
[   82.778741] 5c80: cc1a9000 bf377548 00000001 00000000 cc326e88 cc326e94 cc326ea0 cc326eac
[   82.787341] 5ca0: cc326e08 bf3b8fa8 00000001 cc26ca20 cc1a9068 cc1a9000 bf3b8d34 cc26ca00
[   82.795941] 5cc0: 00000000 bf3b8fa8 00000000 bf019bf4 bf019a80 c0718160 cc26ca20 c070898c
[   82.804541] 5ce0: 00000000 bf3b8d34 00000012 cc326c00 00000001 c02d88d4 c02d8a30 00000000
[   82.813143] 5d00: cc26ca20 c02d8a30 cc1a9068 00000000 bf026e24 c02d7134 cd3634dc cc4232b4
[   82.821744] 5d20: cc26ca20 cc26ca54 bf026e3c c02d87a8 cc26ca28 cc26ca20 bf026e3c c02d7e04
[   82.830345] 5d40: cc26ca28 00000000 cc26ca20 c02d6408 cd555d78 00000000 00000010 00000000
[   82.838945] 5d60: 00000000 cc26ca00 00000000 cc326c50 cc1a9068 cc1a9000 cc2a55c8 bf0182e0
[   82.847544] 5d80: 00000001 00000000 00000000 00000000 00001388 c010b8dc 00000001 00000001
[   82.856144] 5da0: cc326c50 cc2c9e40 00000001 bf026e3c bf026f40 00000001 cc326c00 cc2c9e40
[   82.864744] 5dc0: 00000001 cd7aa000 bf027579 cc326c4c 00000012 cc1a9000 00000001 c070898c
[   82.873345] 5de0: 00000000 bf0274dc 00000012 cd78ba2c cd7aa000 bf02067c bf02063c c0718160
[   82.881945] 5e00: cc1a9068 c02d88d4 c02d8a30 00000000 cc1a9068 c02d8a30 cd7bec68 00000000
[   82.890545] 5e20: bf026cc0 c02d7134 cd3634dc cd496ab4 cc1a9068 cc1a909c bf026e3c c02d87a8
[   82.899145] 5e40: cc1a9070 cc1a9068 bf026e3c c02d7e04 cc1a9070 00000000 cc1a9068 c02d6408
[   82.907746] 5e60: 393831b0 0000313a 000001b2 cd78ba2c cd7aa000 cc1a9000 cc1a9068 cd7bec00
[   82.916346] 5e80: 00000001 000001b4 000001b2 bf00f8a4 00000003 00000000 bf022fdc cc1a9000
[   82.924946] 5ea0: cc1a9000 00000000 cd7aa018 00000000 00000001 00000000 cc1a9000 00000000
[   82.933549] 5ec0: cd7aa018 cd7bec00 00000001 bf011c48 00000000 00002e7b 001e8673 cd7bec9c
[   82.942149] 5ee0: cd78bc00 cd78ba00 bf026ce8 cd78bc20 cd7bec00 00000000 cd78ba7d 00000064
[   82.950749] 5f00: cd7aa000 cd78ba08 00000000 00000011 cd78bc20 c005acf4 bf0276e0 c069a068
[   82.959351] 5f20: c0711130 00000000 c055fdcc 0101adb0 cd370001 00000000 cd370100 c0063668
[   82.967952] 5f40: cd555f40 cd555f40 00000000 cd582e00 00000000 00000000 bf011160 00000000
[   82.976552] 5f60: 00000000 00000000 00000000 c0057bb8 0000000d 00000000 cd582ec0 00000000
[   82.985152] 5f80: 00000000 cd555f84 cd555f84 00000000 cd555f90 cd555f90 cd555fac cd582e00
[   82.993752] 5fa0: c0057ad8 00000000 00000000 c0008d18 00000000 00000000 00000000 00000000
[   83.002352] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   83.010951] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[   83.019559] Code: e4805004 e4806004 e4807004 e4808004 (e480e004)
[   83.026071] ---[ end trace f6c46342573707aa ]---
`

antismap, 2016-08-01:
I'm attaching more debug messages

`[   54.960983] RTL871X:  [0x00100000,5]+rtw_drv_entry
[   54.961031] RTL871X: rtl8812au driver version=v4.2.2_7502.20130517
[   54.961300] RTL871X:  [0x00100000,5]+rtw_drv_init
[   54.961335] RTL871X:
usb_endpoint_descriptor(0):
[   54.961362] RTL871X: bLength=7
[   54.961376] RTL871X: bDescriptorType=5
[   54.961391] RTL871X: bEndpointAddress=81
[   54.961405] RTL871X: wMaxPacketSize=512
[   54.961420] RTL871X: bInterval=0
[   54.961434] RTL871X: RT_usb_endpoint_is_bulk_in = 1
[   54.961450] RTL871X:
usb_endpoint_descriptor(1):
[   54.961470] RTL871X: bLength=7
[   54.961484] RTL871X: bDescriptorType=5
[   54.961498] RTL871X: bEndpointAddress=2
[   54.961512] RTL871X: wMaxPacketSize=512
[   54.961526] RTL871X: bInterval=0
[   54.961541] RTL871X: RT_usb_endpoint_is_bulk_out = 2
[   54.961556] RTL871X:
usb_endpoint_descriptor(2):
[   54.961576] RTL871X: bLength=7
[   54.961590] RTL871X: bDescriptorType=5
[   54.961603] RTL871X: bEndpointAddress=3
[   54.961617] RTL871X: wMaxPacketSize=512
[   54.961631] RTL871X: bInterval=0
[   54.961645] RTL871X: RT_usb_endpoint_is_bulk_out = 3
[   54.961660] RTL871X:
usb_endpoint_descriptor(3):
[   54.961680] RTL871X: bLength=7
[   54.961694] RTL871X: bDescriptorType=5
[   54.961708] RTL871X: bEndpointAddress=4
[   54.961721] RTL871X: wMaxPacketSize=512
[   54.961735] RTL871X: bInterval=0
[   54.961749] RTL871X: RT_usb_endpoint_is_bulk_out = 4
[   54.961764] RTL871X:
usb_endpoint_descriptor(4):
[   54.961784] RTL871X: bLength=7
[   54.961798] RTL871X: bDescriptorType=5
[   54.961812] RTL871X: bEndpointAddress=85
[   54.961826] RTL871X: wMaxPacketSize=64
[   54.961840] RTL871X: bInterval=1
[   54.961855] RTL871X: RT_usb_endpoint_is_int_in = 5, Interval = 1
[   54.961874] RTL871X: nr_endpoint=5, in_num=2, out_num=3

[   54.961895] RTL871X: USB_SPEED_HIGH
[   54.962168] RTL871X: CHIP TYPE: RTL8812
[   54.962270] RTL871X: register rtw_netdev_ops to netdev_ops
[   54.962707] RTL871X: ReadChipVersion8812A SYS_CFG(0xF0)=0x04411135
[   54.962822] RTL871X: Chip Version Info: CHIP_8812_Normal_Chip_TSMC_C_CUT_2T2R_RomVer(0)
[   54.962844] RTL871X: RF_Type is 2!!
[   54.962871] RTL871X: _ConfigChipOutEP_8812 OutEpQueueSel(0x07), OutEpNumber(3)
[   54.962889] RTL871X: ====> ReadAdapterInfo8812AU
[   54.963010] RTL871X: Boot from EFUSE, Autoload OK !
[   55.104261] RTL871X: EEPROM ID=0x8129
[   55.104363] RTL871X: VID = 0x7392, PID = 0xA822
[   55.104386] RTL871X: Customer ID: 0x00, SubCustomer ID: 0xCD
[   55.104418] RTL871X: hal_ReadMACAddress_8812AU MAC Address from EFUSE = 74:da:38:1f:47:70
[   55.104470] RTL871X: ==>mpt_GetChnlGroup8812A in 5G, but Channel 163 in Group not found
[   55.104507] RTL871X: ==>mpt_GetChnlGroup8812A in 5G, but Channel 163 in Group not found
[   55.104541] RTL871X: ==>mpt_GetChnlGroup8812A in 5G, but Channel 163 in Group not found
[   55.104575] RTL871X: ==>mpt_GetChnlGroup8812A in 5G, but Channel 163 in Group not found
[   55.104600] RTL871X: EEPROMRegulatory = 0x1 TxPwrCalibrateRate=0x0
[   55.104616] RTL871X: Board Type: 0x 0
[   55.104634] RTL871X: mlmepriv.ChannelPlan = 0x26
[   55.104650] RTL871X: CrystalCap: 0x1e
[   55.104664] RTL871X: ThermalMeter = 0x28
[   55.104682] RTL871X: SWAS: bHwAntDiv = 0, TRxAntDivType = 3
[... 7 lines cut ...]
[   55.104821] RTL871X: Customer ID: 0x 0
[   55.109113] RTL871X: ReadAdapterInfo8812AU <====
[   55.109330] RTL871X: init_channel_set ChannelPlan ID 26 Chan num:32
[   55.110045] RTL871X: can't get autopm:
[   55.110062] RTL871X: rtw_macaddr_cfg MAC Address  = 74:da:38:1f:47:70
[   55.110073] RTL871X: bDriverStopped:1, bSurpriseRemoved:0, bup:0, hw_init_completed:0
[   55.110211] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   55.118765] pgd = cc23c000
[   55.121602] [00000000] *pgd=8c361831, *pte=00000000, *ppte=00000000
[   55.128211] Internal error: Oops: 817 [#1] ARM
[   55.132874] Modules linked in: 8812au(+) option iptable_nat usb_wwan sierra pppoe nf_nat_ipv4 nf_conntrack_netlink nf_conntrack_ipv6 nf_conntrack_ipv4 mcs7830 ipt_MASQUERADE ftdi_sio cdc_ether xt_time xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda pptp pppox ppp_mppe ppp_async nfnetlink nf_nat_irc nf_nat_ftp nf_nat nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_irc nf_conntrack_ftp nf_conntrack iptable_raw iptable_mangle iptable_filter ipt_REJECT ip_tables fuse sch_teql sch_tbf sch_sfq sch_red sch_prio sch_htb sch_gred sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_police
[   55.207858]  act_ipt act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress musb_am335x musb_dsps musb_hdrc udc_core phy_am335x phy_am335x_control phy_generic ip6t_REJECT ip6table_raw ip6table_mangle ip6table_filter ip6_tables swconfig pppoatm ppp_generic slhc nfsd ip_gre gre tun cifs dm_crypt dm_mirror dm_region_hash dm_log dm_mod br2684 atm nls_utf8 nls_cp850 cryptosoft cryptodev ocf md4 usb_storage ehci_platform ehci_hcd usbcore usb_common exportfs mii
[   55.252838] CPU: 0 PID: 3925 Comm: insmod Not tainted 3.14.26 #17
[   55.259223] task: cd1cf500 ti: cc1be000 task.ti: cc1be000
[   55.264899] PC is at memcpy+0xb4/0x330
[   55.268849] LR is at 0x1f38da74
[   55.272143] pc : [<c000d1d4>]    lr : [<1f38da74>]    psr: 000b0013
sp : cc1bfc9c  ip : 0000001c  fp : d1357000
[   55.284179] r10: 000040d0  r9 : 00004180  r8 : cc2f5400
[   55.289655] r7 : cc2f5400  r6 : 00000000  r5 : cc36e000  r4 : d1357000
[   55.296497] r3 : 00000000  r2 : ffffffe6  r1 : d135a1ec  r0 : 00000000
[   55.303343] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   55.310824] Control: 10c5387d  Table: 8c23c019  DAC: 00000015
[   55.316850] Process insmod (pid: 3925, stack limit = 0xcc1be238)
`

antismap, 2016-08-01:
Ok so the problem comes from os_dep/linux/os_infs.c line 2420 : 
In the function 
`static int _rtw_drv_register_netdev(_adapter *padapter, char *name)`
we call 
`int register_netdev(struct net_device *dev);`

Problem : some of the structure is not properly allocated and this makes the whole thing crash, probably when trying to memcpy something into it.

This looks nontrivial because the structure is [not small]((http://lxr.free-electrons.com/source/include/linux/netdevice.h?v=3.14#L1160) and I don't know was what wrongly initialized.

antismap, 2016-08-01:
actually the crash already happens at this line : 
`_rtw_memcpy(pnetdev->dev_addr, padapter->eeprompriv.mac_addr, ETH_ALEN);
`
if malloc pnetdev->dev_addr just before the call, the crash happens at register_netdev.

antismap, 2016-08-02:
After reading this about a [similar issue](https://github.com/lwfinger/rtl8188eu/issues/110) on a similar driver, I'm starting to think that this could be a structure alignment issue because I'm on ARM ?

Swebb75, 2017-01-26:
Were you able to figure out a resolution on this problem?  I'm running into a similar error (kernel panic, EMI? error message, de-referencing invalid pointer) with a USB camera.  A difference is that it does work for a while, and only crashes after the port gets disabled (EMI? error)

[30715.551953@0] hub 1-0:1.0: port 1 disabled by hub (EMI?), re-enabling...
[30715.553037@0] usb 1-1: USB disconnect, device number 2
[30715.558142@0] usb 1-1.3: USB disconnect, device number 3
Jan 25 22:11:52 localhost kernel: [30715.553037@0] usb 1-1: USB disconnect, device number 2
Jan 25 22:11:52 localhost kernel: [30715.558142@0] usb 1-1.3: USB disconnect, device number 3
[30715.753598@0] usb 1-1.4: USB disconnect, device number 4
[30715.753641@0] usb 1-1.4.1: USB disconnect, device number 5
Jan 25 22:11:52 localhost kernel: [30715.753598@0] usb 1-1.4: USB disconnect, device number 4
Jan 25 22:11:52 localhost kernel: [30715.753641@0] usb 1-1.4.1: USB disconnect, device number 5
[30725.411938@3] WARN::hc_xfer_timeout:2719: hc_xfer_timeout: timeout on channel 14
[30725.413697@3] WARN::hc_xfer_timeout:2721:    start_hcchar_val 0x01748c00
[30725.420211@3] WARN::hc_xfer_timeout:2726:    chn-14,ep1-IN:type:1,speed:2,len:3072,addr5
[30725.420211@3]
Jan 25 22:12:02 localhost kernel: [30725.411938@3] WARN::hc_xfer_timeout:2719: hc_xfer_timeout: timeout on channel 14
Jan 25 22:12:02 localhost kernel: [30725.413697@3] WARN::hc_xfer_timeout:2721:  start_hcchar_val 0x01748c00
Jan 25 22:12:02 localhost kernel: [30725.420211@3] WARN::hc_xfer_timeout:2726:  chn-14,ep1-IN:type:1,speed:2,len:3072,addr5
Jan 25 22:12:02 localhost kernel: [30725.420211@3]
[30725.931940@1] Indeed it is in host mode hprt0 = 00001501
Jan 25 22:12:02 localhost kernel: [30725.931940@1] Indeed it is in host mode hprt0 = 00001501
[30725.992336@3] Unable to handle kernel NULL pointer dereference at virtual address 00000036
[30725.994961@3] pgd = c0004000
[30725.997825@3] [00000036] *pgd=00000000
[30726.001563@3] Internal error: Oops: 17 [#1] PREEMPT SMP ARM
[30726.007117@3] Modules linked in: nls_cp437 snd_usb_audio uvcvideo snd_hwdep snd_usbmidi_lib snd_rawmidi w1_gpio wire videobuf2_vmalloc snd_seq_device
[30726.020494@3] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 3.10.72 #12
[30726.026744@3] task: ef274780 ti: ef2a2000 task.ti: ef2a2000
[30726.032308@3] PC is at handle_hc_xacterr_intr+0x68/0x17c
[30726.037600@3] LR is at dwc_otg_hcd_handle_hc_n_intr+0x308/0x894
[30726.043505@3] pc : [<c040bb64>]    lr : [<c040bf80>]    psr: 60070193
[30726.043505@3] sp : ef2a3e60  ip : c0a9f3b0  fp : c59b2e88
[30726.055314@3] r10: c0a9f3b0  r9 : fe0806c0  r8 : ef501b80
[30726.060698@3] r7 : fe0806c0  r6 : ef501b80  r5 : c5694600  r4 : c59b2e88
[30726.067385@3] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : c5694600
[30726.074072@3] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[30726.081626@3] Control: 10c5387d  Table: 2e80004a  DAC: 00000015

I'm on 3.10.72, so I'm going to try a newer version of the kernel to see if it has been fixed.  Any input would be appreciated.

dismirlian, 2017-05-24:
Hi, I'm having the same issue:

Unable to handle kernel NULL pointer dereference at virtual address 00000004
pgd = c5918000
[00000004] *pgd=00000000
Internal error: Oops: 805 [#1] ARM
Modules linked in: maper_linux(O) snd_atmel_soc_pcm1864 snd_soc_atmel_ssc_dai snd_soc_atmel_pcm_dma 8192cu(O+) snd_soc_pcm186x_i2c snd_soc_pcm186x
CPU: 0 PID: 809 Comm: systemd-udevd Tainted: G           O    4.1.0-linux4sam_5.3-00053-gaa504b4-dirty #37
Hardware name: Atmel SAMA5
task: c5238400 ti: c5914000 task.ti: c5914000
PC is at memcpy+0x1c0/0x330
LR is at 0x3d8e
pc : [<c01c49e0>]    lr : [<00003d8e>]    psr: 60040113
sp : c5915ccc  ip : 00000000  fp : bf074afc
r10: c43a9580  r9 : 000038b0  r8 : 00000000
r7 : c43a9580  r6 : d09f8111  r5 : c44dd800  r4 : d09f5000
r3 : 8eb9fd6c  r2 : ffffffe6  r1 : d09f8118  r0 : 00000004
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 10c53c7d  Table: 25918059  DAC: 00000015
Process systemd-udevd (pid: 809, stack limit = 0xc5914208)
Stack: (0xc5915ccc to 0xc5916000)
5cc0:                            00000004 d09f5000 bf05c160 00000000 bf05e180
5ce0: 00000000 c0434fc4 d09f8111 c5915cfc 00000000 bf05fc3c bf07522c d09f5000
5d00: 00000000 d09f5000 c43a9580 c44a0a00 c4453fd0 000000c0 c42d1068 00000004
5d20: c43a9590 bf0600b4 c4451c08 c42d1000 bf075284 bf0752a4 bf0752b0 c42d1068
5d40: bf08cc18 c44a0a20 c42d1000 00000000 bf08b9d4 c44a0a00 00000000 bf08c244
5d60: 0000adc7 c028fea0 c061c68c c44a0a20 00000000 bf08b9d4 00000004 bf08cc18
5d80: 00000000 c0228dfc c44a0a20 bf08b9d4 c44a0a54 00000000 bf08cbd0 c0228f2c
5da0: 00000000 bf08b9d4 c0228ea0 c022768c c416214c c430e330 bf08b9d4 c529a480
5dc0: c05f11cc c0228528 bf075520 bf0a3000 bf08b9d4 bf08b9d4 bf0754f8 bf08b9d4
5de0: c05f11cc c0229534 bf08b9cc bf08b9a4 bf0754f8 c028ebe8 c05d9a58 c05d9a58
5e00: c6198e80 c6028fc0 bf0a3000 bf0a3088 00000000 c00095bc c05feae8 c05feb04
5e20: 00000000 c05fee70 00000000 cfdb9000 00100100 00200200 ffffffff 00000001
5e40: c05feee4 c05feef0 c01c83c0 0000000f c05d99c0 cfe99e60 00000000 00020000
5e60: cfdda000 00000000 bf08cbd0 00000001 c5df8c80 c6028fc0 00000001 bf08cc18
5e80: c00584b4 c04350c0 d0b01000 c008b8b4 c5915f58 c5915f58 00000001 bf08cbd0
5ea0: c6028fc0 c005a8b8 bf08cbdc 00007fff c0058640 bf000000 c0058528 00000124
5ec0: 00000000 bf08cd18 b6f11efc d0b852c4 00000000 bf08cbdc 00000000 c0058364
5ee0: d0b01000 000c20ab 00094a9b 00000000 0000000d 00000000 00000000 00000000
5f00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
5f20: 00000000 00000000 00000000 00000000 00000618 00000000 00000007 b6f11efc
5f40: 0000017b c000f3a4 c5914000 00000000 7f5dd538 c005b048 d0b01000 000c20ab
5f60: d0b84dc4 d0b84c6f d0bb82e4 000858e8 0008f448 00000000 00000000 00000000
5f80: 00000020 00000021 00000016 00000000 0000000a 00000000 00000000 7f5dd538
5fa0: 00000000 c000f220 00000000 7f5dd538 00000007 b6f11efc 00000000 00000000
5fc0: 00000000 7f5dd538 00000000 0000017b 00000000 7f5cbc10 7f5dc054 7f5dd538
5fe0: bee01a10 bee01a00 b6f0ac78 b6e3d880 60070010 00000007 ffffffff ffffffff
[<c01c49e0>] (memcpy) from [<bf05c160>] (_rtw_memcpy+0x8/0xc [8192cu])
[<bf05c160>] (_rtw_memcpy [8192cu]) from [<bf05e180>] (rtw_drv_register_netdev+0xc8/0x198 [8192cu])
[<bf05e180>] (rtw_drv_register_netdev [8192cu]) from [<bf0600b4>] (rtw_drv_init+0x400/0x59c [8192cu])
[<bf0600b4>] (rtw_drv_init [8192cu]) from [<c028fea0>] (usb_probe_interface+0x154/0x1f8)
[<c028fea0>] (usb_probe_interface) from [<c0228dfc>] (driver_probe_device+0x1c0/0x264)
[<c0228dfc>] (driver_probe_device) from [<c0228f2c>] (__driver_attach+0x8c/0x90)
[<c0228f2c>] (__driver_attach) from [<c022768c>] (bus_for_each_dev+0x54/0x88)
[<c022768c>] (bus_for_each_dev) from [<c0228528>] (bus_add_driver+0xd4/0x1d0)
[<c0228528>] (bus_add_driver) from [<c0229534>] (driver_register+0x78/0xf4)
[<c0229534>] (driver_register) from [<c028ebe8>] (usb_register_driver+0x60/0x128)
[<c028ebe8>] (usb_register_driver) from [<bf0a3088>] (rtw_drv_entry+0x88/0xc0 [8192cu])
[<bf0a3088>] (rtw_drv_entry [8192cu]) from [<c00095bc>] (do_one_initcall+0x80/0x1d0)
[<c00095bc>] (do_one_initcall) from [<c04350c0>] (do_init_module+0x58/0x1a8)
[<c04350c0>] (do_init_module) from [<c005a8b8>] (load_module+0x1570/0x1b80)
[<c005a8b8>] (load_module) from [<c005b048>] (SyS_finit_module+0x64/0x70)
[<c005b048>] (SyS_finit_module) from [<c000f220>] (ret_fast_syscall+0x0/0x3c)
Code: e1a0342e e491e004 e25cc004 e1833c0e (e4803004)
---[ end trace 02852e685953b5e7 ]---

I'm getting dev_addr=00000004 at function rtw_alloc_etherdev_with_old_priv; this comes directly from alloc_etherdev_mq, which _should_ initialize dev_addr correclty. Any clues?

Kernel version: 4.1.0

Thank you!!