gnab/rtl8812au #105: Explanation for DKMS

goupil35000, 2017-03-08:
Hi,
You told on your 1st page for this driver that for dkms version, we have to do this:
>> The driver source mus be copied to /usr/src/8812au-4.2.2

Is-it the directory rtl8812au-master we have to copy or the contents of this directory ?

Thanks

ManuGithubSteam, 2017-05-27:
Hi

This depens on what linux distro you use. Check first, if there is another name present.

in the end it does not matter how you name it but you need to be consistent for the install process i asume.

If you want to copy it, only copy the contents!

Have fun

ManuGithubSteam, 2017-07-09:
If this solved your isse, please close it.

RICCIARDI-Adrien, 2020-09-28:
Do you think this issue could be closed now that DKMS installation is handled by makefile rules ?

closed 2020-11-20 by goupil35000