gnab/rtl8812au #106: Complied and installed with no error msg. BUT cannot find wifi icon.

Nina0109, 2017-04-27:
Hi, my wireless adapter is TOTOLINK A2000UA, chip is rtl8812au.
My host os is centos 7, version 3.10.0, x86_64
_Linux localhost.localdomain 3.10.0-514.16.1.el7.x86_64 #1 SMP Wed Apr 12 15:04:24 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux_
After download your driver, I do like this:
    make 
    sudo make install
    sudo modprobe 8812au
I didn't see any error. BUT, after all these, I cannot see the wifi icon, the indicator light on the Adapter is also not on.
Can someone tell me what happened? What should I do next?

pbeejelly, 2017-04-30:
I have the same problem as well, but my wifi indicator light shines a static light.
I am currently using Xubuntu 16.04 LTS and installed everything with no error, but I do not see a wifi icon available on my desktop during&after restarting.

I installed the dkms on my desktop. 
Copied the files into usr/src/
Ran the commands:
$ sudo dkms add -m 8812au -v 4.2.2
$ sudo dkms build -m 8812au -v 4.2.2
$ sudo dkms install -m 8812au -v 4.2.2

and checked status

Nina0109, 2017-05-01:
My wifi adapter can work now. Although I still cannot see the wifi icon.
I run following command in the terminal:
iwlist <adapter name> scanning(it shows available APs)
If the ap is not encrypted or encrypted by WEP, we can connect it by iwconfig
iwconfig <adapter name> essid <ap essid> key <ap key>
If it is encrypted by WAP, we should install wpa_supplicant,and write essid and key in its config file.
Use command in wpa supplicant package to connect to wifi.

closed 2017-05-01 by Nina0109