gnab/rtl8812au #111: Driver must be reinstalled on every boot

jimbudarz, 2017-06-04:
I've followed the instructions to rebuild and install on kernel update, but it doesn't work for me (perhaps I'm doing something wrong.)
Installation with install.sh script works beautifully on Ubuntu 17.04 (kernel 4.10.0-21-generic) but when I reboot my system the device is missing until I run the installation script again.  After following the DKMS instructions, $ sudo dkms status gives the following result:

8812au, 4.2.2, 4.10.0-21-generic, x86_64: installed (WARNING! Diff between built and installed module!)

I'm not sure if these issues are related.

Thanks in advance!

rushton3179, 2017-06-17:
I have had a similar issue but I found unplugging and plugging in the wifi device fixed it. This was annoying so I wrote this script to get around the problem (well you need to run it here and there but you could have it set to run on log in)

https://github.com/rushton3179/rtl8812au-installer

jimbudarz, 2017-06-26:
Thanks, rushton3179, but unplugging and plugging in my USB-AC56 doesn't have any effect.
I did find another project that works (https://github.com/abperiasamy/rtl8812AU_8821AU_linux).

ManuGithubSteam, 2017-07-09:
Then please close you issue :-)

jimbudarz, 2017-07-10:
Issue closed without resolution.

closed 2017-07-10 by jimbudarz

kyle-sama, 2018-07-30:
This happens to me. need to reinstall driver every time I reboot.