gnab/rtl8812au #112: tried for Banana Pi

aueb, 2017-06-05:
wanted compile it for Banana Pi got error message

My Device: 
" Bus 002 Device 002: ID 7392:a812 Edimax Technology Co., Ltd "

..: make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.4.103/build M=/media/4464C2B264C2A5D0/daten/sw/Treiber/rtl8812au-master  modules
make[1]: Entering directory '/lib/modules/3.4.103/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/3.4.103/build'
Makefile:1052: recipe for target 'modules' failed
make: *** [modules] Error 2

Line 1052

1051: modules:
1052:	$(MAKE) ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) -C $(KSRC) M=$(shell pwd)  modules
1053:

is it correct to use
CONFIG_PLATFORM_ARM_RPI = y for BananaPi?

Thank you

igorpecovnik, 2017-06-05:
On armbian you only need to run make arch=arm but i am not sure this driver work on legacy kernel while it works with mainline. External powering is also must have.

Jim972, 2017-06-13:
I was faced to the same issue, I put the step-by-step hope this will help you

1. Prepare raspbin for bananapi OS SD card
    Image: Raspbian_For_BananaPi_v1412.img
    MiniTool Partition -> Clean all partitions & rebuild one FAT32 partition on SD card.
    Win32DiskImager -> Write BananaPi image file to SD card.

2. Boot & init Banana pi

a. Boot and look for BananaPi ip. Then ssh to BananaPi as bananapi account, password is [bananapi].
b. Run [rasp-config] to [Expand Filesystem]. Reboot BananaPi to apply change!
c. Login as bananapi, run [passwd] to change password of bananapi account.
d. Run [sudo apt-get update].

3. Find & get Module.symvers for kernel 3.4.111-bananian from internet!
a. find on the web or
b. apt-get install linux-header-$(uname -r)

4.Compile kernel
$ sudo -s
# apt-get update
# uname -r  
# apt-get install build-essential u-boot-tools uboot-mkimage
# apt-get install libusb-1.0-0 libusb-1.0-0-dev git wget fakeroot kernel-package zlib1g-dev libncurses5-dev

# cd /usr/src
# git clone https://github.com/Bananian/linux-bananapi
# ln -s /usr/src/linux-bananapi /lib/modules/3.4.111-bananian/build

# cd linux-bananapi
# make mrproper
# gzip -dc /proc/config.gz > .config
# cp .config .config.org
# make modules_prepare
# cp ~bananapi//Module.symvers .

5.Compile Netgear wifi driver

# cd
# git clone https://github.com/gnab/rtl8812au.git
# cd rtl8812au

Change Makefile and put this value for
a. CONFIG_PLATFORM_I386_PC = n
b. CONFIG_PLATFORM_ARM_PI = y

# chmod +x install.sh
#./install.sh

6. Fix wireless setting.
# vi /etc/network/interfaces
   Change interfaces file like ...
      auto lo iface lo inet loopback
      iface eth0 inet dhcp
      auto wlan0
      allow-hotplug wlan0
      iface wlan0 inet dhcp
      wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
      iface default inet dhcp

# vi /etc/wpa_supplicant/wpa_supplicant.conf
   Change wpa_supplicant.conf like ...
      ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
      update_config=1
      network={
          ssid="YOURWIFINAME"
          key_mgmt=WPA-PSK
          psk="YOURPASSWORD"
      }

# reboot

igorpecovnik, 2017-06-13:
@Jim972 Now repeat this on https://www.armbian.com/banana-pi and how-to will be few lines + you end up with supported, secure and up-to-date kernel.