gnab/rtl8812au #113: Kernel 4.11.9 does not compile...

ManuGithubSteam, 2017-07-08:
Hi

After upgrading to kernel 4.11.9 in Arch Linux i get this compile error:

/var/lib/dkms/rtl8812au_rtl8821au/4.3.20.r16.d716b38/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_add_monitor_if’:
/var/lib/dkms/rtl8812au_rtl8821au/4.3.20.r16.d716b38/build/os_dep/linux/ioctl_cfg80211.c:3952:12: error: ‘struct net_device’ has no member named ‘destructor’; did you mean ‘priv_destructor’?
  mon_ndev->destructor = rtw_ndev_destructor;
            ^~~~~~~~~~
            priv_destructor
make[2]: *** [scripts/Makefile.build:295: /var/lib/dkms/rtl8812au_rtl8821au/4.3.20.r16.d716b38/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1492: _module_/var/lib/dkms/rtl8812au_rtl8821au/4.3.20.r16.d716b38/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.11.9-1-ARCH/build'
make: *** [Makefile:1670: modules] Error 2

How can i fix this ? Is it really so simple in renaming the destructor to priv_destructor?

Obviously the module does not build and so no wifi connections work.

Thanks!

ol3gart, 2017-07-08:
Have the same problem - after update arch not see my TP-LINK T4UH anymore.

ManuGithubSteam, 2017-07-09:
Did you find a solution ?

ManuGithubSteam, 2017-07-09:
Found the fix here: https://github.com/Grawp/rtl8812au_rtl8821au/pull/46

Is is possible to clone the pull request into this repo so AUR package gets updated ?

Thanks

phouverneyuff, 2017-07-09:
Someone can build with this change that I made and to test if works?

ManuGithubSteam, 2017-07-10:
Hi phouverneyuff,

i Just tried to build it and unfortunately it does not.
The error log is large so i'm not posting it here.

You can see it here: https://pastebin.com/9Y8tuUW6

Hope this helps.
Thanks for your hard work!

phouverneyuff, 2017-07-10:
@ManuGithubSteam I only can see this more later today. If anyone have time, can do it too.

PR #114 (2017-10-02): Fix build in Kernel 4.11.9
This PR is based on https://github.com/Grawp/rtl8812au_rtl8821au/pull/46/commits/58fc45a4511b8b9d6b52813168e3eee657517b1f to resolve #113
commit f8e09835 (2017-10-02): Fix build in Kernel 4.11.9

This PR is based on https://github.com/Grawp/rtl8812au_rtl8821au/pull/46/commits/58fc45a4511b8b9d6b52813168e3eee657517b1f to resolve #113
closed 2017-10-02 by gnab