gnab/rtl8812au #115: Linksys WUSB6300 shows as wired ethernet on CentOS 7 1611 Final (kernel 3.10.0-514.el7.x86_64)

Spork-Schivago, 2017-08-15:
Hello,

I'm having issues with this module.   I'm running a fresh install of CentOS 7 1611 Final and I compile the module, insert it into the kernel using insmod, insert the USB wireless NIC (a Linksys WUSB6300 with the RealTek 8812AU chipset), and the device appears in Gnome as a wired ethernet card, not a wireless NIC.

Any suggestions on how to fix this?   I'm not sure if it's something with the module or CentOS 7.   But from reading stuff on the internet, it seems a lot of people have this issue when they're using the module with the RealTek 8812AU chipset.

I'm running the kernel that on the installation media, 3.10.0-514.el7.x86_64.   Unfortunately, at this point, I have no way of hooking the machine to update it until I get the wireless NIC working, so I don't know if there's a newer kernel available or not.

I would be down for providing more information, if you need it, and I will test anything you want tested.

Thank you.

codibit, 2017-11-11:
Same here, kernel 3.10.0-693.el7.x86_64.
Compiled from master.

Device is a ¨chinese¨(or not very well known) brand, shows up as RTL8812AU in lsusb (Bus 003 Device 007: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter).

After loading module, appears as a wired network in network manager. Here´s the correponding ip a line:
5: enp0s20u9: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether e8:4e:06:4a:c5:ba brd ff:ff:ff:ff:ff:ff

Let me know if I can add any more info or logs.

killrazor, 2018-02-22:
Same

Device is Linksys AC1200 (model WUSB6300)
CentOS Linux release 7.4.1708 (Core)
3.10.0-693.17.1.el7.x86_64
enp5s0u1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether c8:d7:19:c4:cb:7c  txqueuelen 1000  (Ethernet)

lainme, 2018-11-18:
I have both Centos 7 and Archlinux on the same computer. This problem seems to be Centos 7 specific, since I never have problem with Archlinux.

I  tried to update the kernel in Centos 7 to 4.4 and even 4.19 but with no luck, so it's not related to kernel. I also tried 'lsusb' on both system, and found the device name is not shown in Centos 7, there only have "Bus 005 Device 002: ID 2357:0101".  And here is the output of 'lshw'

  *-network
       description: Wireless interface
       physical id: 1
       bus info: usb@5:2
       logical name: wlan0
       serial: 14:75:90:f9:33:3c
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8812au ip=192.168.0.20 multicast=yes wireless=IEEE 802.11bgn

jtome7, 2020-07-21:
Three years later ... Linksys WUSB6300 Wifi Adapter WORKS via NetworkManage on CentOS 7.8 kernel 3.10.0-1127.13.1.el7.x86_64 

# yum install kmod-rtl8812au.x86_64

# lsmod | grep XX
88XXau               2189305  0 
cfg80211              710770  5 ath,ath9k_common,ath9k,mac80211,88XXau

# lsusb
Bus 003 Device 008: ID 13b1:003f Linksys WUSB6300 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU]

# iwconfig
wlp0s20u5  IEEE 802.11AC  ESSID:"SBG6900AC-DC966-5G"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.745 GHz  Access Point: 5C:E3:0E:96:D7:A0   
          Bit Rate:174 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=83/100  Signal level=36/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

# ifconfig 
wlp0s20u5: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.117  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::bbf5:446d:e3ec:90fd  prefixlen 64  scopeid 0x20<link>
        inet6 2606:a000:810c:9300:9c04:74bc:9909:73d  prefixlen 64  scopeid 0x0<global>
        inet6 2606:a000:810c:9300::6  prefixlen 128  scopeid 0x0<global>
        ether c4:41:1e:5d:7f:98  txqueuelen 1000  (Ethernet)
        RX packets 1480  bytes 999935 (976.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1724  bytes 484480 (473.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

# nmcli con
NAME                     UUID                                  TYPE      DEVICE    
SBG6900AC-DC966-5G       fd0097f7-2c89-4a2b-bb8e-a23e5d197ac2  wifi      wlp0s20u5

# nmcli device show
GENERAL.DEVICE:                         wlp0s20u5
GENERAL.TYPE:                           wifi
GENERAL.HWADDR:                         C4:41:1E:5D:7F:98
GENERAL.MTU:                            1500
GENERAL.STATE:                          100 (connected)
GENERAL.CONNECTION:                     SBG6900AC-DC966-5G
GENERAL.CON-PATH:                       /org/freedesktop/NetworkManager/ActiveConnection/9
IP4.ADDRESS[1]:                         192.168.0.117/24
IP4.GATEWAY:                            192.168.0.1
IP4.ROUTE[1]:                           dst = 0.0.0.0/0, nh = 192.168.0.1, mt = 600
IP4.ROUTE[2]:                           dst = 192.168.0.0/24, nh = 0.0.0.0, mt = 600
IP4.DNS[1]:                             209.18.47.62
IP4.DNS[2]:                             209.18.47.61
IP4.DOMAIN[1]:                          triad.rr.com
IP6.ADDRESS[1]:                         2606:a000:810c:9300:9c04:74bc:9909:73d/64
IP6.ADDRESS[2]:                         2606:a000:810c:9300::6/128
IP6.ADDRESS[3]:                         fe80::bbf5:446d:e3ec:90fd/64
IP6.GATEWAY:                            fe80::5ee3:eff:fe96:d79e
IP6.ROUTE[1]:                           dst = 2606:a000:810c:9300::/56, nh = fe80::5ee3:eff:fe96:d79e, 
IP6.ROUTE[2]:                           dst = 2606:a000:810c:9300::/64, nh = ::, mt = 600
IP6.ROUTE[3]:                           dst = ::/0, nh = fe80::5ee3:eff:fe96:d79e, mt = 600
IP6.ROUTE[4]:                           dst = 2606:a000:810c:9300::6/128, nh = ::, mt = 600
IP6.ROUTE[5]:                           dst = fe80::/64, nh = ::, mt = 600
IP6.ROUTE[6]:                           dst = ff00::/8, nh = ::, mt = 256, table=255

[root@blue wpa_supplicant]# nmcli dev wifi
IN-USE  SSID                 MODE   CHAN  RATE        SIGNAL  BARS  SECURITY 
        SBG6900AC-DC966      Infra  1     195 Mbit/s  67      ▂▄▆_  WPA2     
*       SBG6900AC-DC966-5G   Infra  149   195 Mbit/s  64      ▂▄▆_  WPA2

Another wifi device that I found to work on CentOS Linux 7 is TP-LINK Archer T9UH  AC-1900  USB 3.0 Adapter.

alanfanh, 2020-09-02:
> Three years later ... Linksys WUSB6300 Wifi Adapter WORKS via NetworkManage on CentOS 7.8 kernel 3.10.0-1127.13.1.el7.x86_64
> 
> # yum install kmod-rtl8812au.x86_64
> # lsmod | grep XX
> 88XXau 2189305 0
> cfg80211 710770 5 ath,ath9k_common,ath9k,mac80211,88XXau
> 
> # lsusb
> Bus 003 Device 008: ID 13b1:003f Linksys WUSB6300 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU]
> 
> # iwconfig
> wlp0s20u5 IEEE 802.11AC ESSID:"SBG6900AC-DC966-5G" Nickname:"[WIFI@REALTEK](mailto:WIFI@REALTEK)"
> Mode:Managed Frequency:5.745 GHz Access Point: 5C:E3:0E:96:D7:A0
> Bit Rate:174 Mb/s Sensitivity:0/0
> Retry:off RTS thr:off Fragment thr:off
> Encryption key:****-****-****-****-****-****-****-**** Security mode:open
> Power Management:off
> Link Quality=83/100 Signal level=36/100 Noise level=0/100
> Rx invalid nwid:0 Rx invalid crypt:0 Rx invalid frag:0
> Tx excessive retries:0 Invalid misc:0 Missed beacon:0
> 
> # ifconfig
> wlp0s20u5: flags=4163<UP,BROADCAST,RUNNING,MULTICAST> mtu 1500
> inet 192.168.0.117 netmask 255.255.255.0 broadcast 192.168.0.255
> inet6 fe80::bbf5:446d:e3ec:90fd prefixlen 64 scopeid 0x20
> inet6 2606:a000:810c:9300:9c04:74bc:9909:73d prefixlen 64 scopeid 0x0
> inet6 2606:a000:810c:9300::6 prefixlen 128 scopeid 0x0
> ether c4:41:1e:5d:7f:98 txqueuelen 1000 (Ethernet)
> RX packets 1480 bytes 999935 (976.4 KiB)
> RX errors 0 dropped 0 overruns 0 frame 0
> TX packets 1724 bytes 484480 (473.1 KiB)
> TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0
> 
> # nmcli con
> NAME UUID TYPE DEVICE
> SBG6900AC-DC966-5G fd0097f7-2c89-4a2b-bb8e-a23e5d197ac2 wifi wlp0s20u5
> 
> # nmcli device show
> GENERAL.DEVICE: wlp0s20u5
> GENERAL.TYPE: wifi
> GENERAL.HWADDR: C4:41:1E:5D:7F:98
> GENERAL.MTU: 1500
> GENERAL.STATE: 100 (connected)
> GENERAL.CONNECTION: SBG6900AC-DC966-5G
> GENERAL.CON-PATH: /org/freedesktop/NetworkManager/ActiveConnection/9
> IP4.ADDRESS[1]: 192.168.0.117/24
> IP4.GATEWAY: 192.168.0.1
> IP4.ROUTE[1]: dst = 0.0.0.0/0, nh = 192.168.0.1, mt = 600
> IP4.ROUTE[2]: dst = 192.168.0.0/24, nh = 0.0.0.0, mt = 600
> IP4.DNS[1]: 209.18.47.62
> IP4.DNS[2]: 209.18.47.61
> IP4.DOMAIN[1]: triad.rr.com
> IP6.ADDRESS[1]: 2606:a000:810c:9300:9c04:74bc:9909:73d/64
> IP6.ADDRESS[2]: 2606:a000:810c:9300::6/128
> IP6.ADDRESS[3]: fe80::bbf5:446d:e3ec:90fd/64
> IP6.GATEWAY: fe80::5ee3:eff:fe96:d79e
> IP6.ROUTE[1]: dst = 2606:a000:810c:9300::/56, nh = fe80::5ee3:eff:fe96:d79e,
> IP6.ROUTE[2]: dst = 2606:a000:810c:9300::/64, nh = ::, mt = 600
> IP6.ROUTE[3]: dst = ::/0, nh = fe80::5ee3:eff:fe96:d79e, mt = 600
> IP6.ROUTE[4]: dst = 2606:a000:810c:9300::6/128, nh = ::, mt = 600
> IP6.ROUTE[5]: dst = fe80::/64, nh = ::, mt = 600
> IP6.ROUTE[6]: dst = ff00::/8, nh = ::, mt = 256, table=255
> 
> [root@blue wpa_supplicant]# nmcli dev wifi
> IN-USE SSID MODE CHAN RATE SIGNAL BARS SECURITY
> SBG6900AC-DC966 Infra 1 195 Mbit/s 67 ▂▄▆_ WPA2
> 
> * ```
>     SBG6900AC-DC966-5G   Infra  149   195 Mbit/s  64      ▂▄▆_  WPA2
>   ```
[... 63 lines cut ...]
> IP6.ROUTE[5]: dst = fe80::/64, nh = ::, mt = 600
> IP6.ROUTE[6]: dst = ff00::/8, nh = ::, mt = 256, table=255
> 
> [root@blue wpa_supplicant]# nmcli dev wifi
> IN-USE SSID MODE CHAN RATE SIGNAL BARS SECURITY
> SBG6900AC-DC966 Infra 1 195 Mbit/s 67 ▂▄▆_ WPA2
> 
> * ```
>     SBG6900AC-DC966-5G   Infra  149   195 Mbit/s  64      ▂▄▆_  WPA2
>   ```
> 
> Another wifi device that I found to work on CentOS Linux 7 is TP-LINK Archer T9UH AC-1900 USB 3.0 Adapter.

hi ，I use the command in Centos7.7 kernel 3.10.0-1127.19.1.el7.x86_64  , but it‘s doesn't work.
Here is the information
-----------------------
yum install kmod-rtl8812au.x86_64
Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: mirrors.aliyun.com
 * epel: mirrors.thzhost.com
 * extras: mirrors.ustc.edu.cn
 * updates: mirrors.ustc.edu.cn
No package kmod-rtl8812au.x86_64 available.
Error: Nothing to do