gnab/rtl8812au #119: nolinked power save enter <-> leave

pnpavlov, 2017-08-30:
Hi,

I tried two approaches to configure my USB stick on  Fedora 26 running
```
# uname -rvi
4.12.8-300.fc26.x86_64 #1 SMP Thu Aug 17 15:30:20 UTC 2017 x86_64
```
I tried doing git clone https://github.com/gnab/rtl8812au.git and then 
```
# Option A 
cd rtl8812au/
make
cp 8812au.ko /lib/modules/$(uname -r)/kernel/drivers/net/wireless
depmod
insmod 8812au.ko
```
Then cleaned up and did
```
# Option B
cp -rf rtl8812au/ /usr/src/8812au-4.2.2
dkms add -m 8812au -v 4.2.2
dkms build -m 8812au -v 4.2.2
dkms install -m 8812au -v 4.2.2
```
Both time compile time I'm good and module is loaded. Then the USB stick blinks few times when I turn on/off the wifi from the gnome manager.
Unfortunately I can't see any network and I can't connect to "hidden network" using my SSID & Pass. 
dmesg is also not very helpfull...
```
# dmesg | grep RTL
[   37.627024] RTL871X: set bssid:00:00:00:00:00:00
[   37.627031] RTL871X: set ssid [g\xc6isQ\xffJ\xec)ͺ\xab\xf2\xfb\xe3F|\xc2T\xf8\xe8\xe7\x8dvZ.c3\x9fɚ\xb6\x97\xbc\x93\xe5\xbdI] fw_state=0x00000008
[   43.747239] RTL871X: nolinked power save enter
[   64.972382] RTL871X: nolinked power save leave
[   71.180333] RTL871X: nolinked power save enter
[   97.832065] RTL871X: nolinked power save leave
[  103.934330] RTL871X: nolinked power save enter
[  108.503008] RTL871X: nolinked power save leave
[  114.622680] RTL871X: nolinked power save enter
[  119.871456] RTL871X: nolinked power save leave
[  126.016077] RTL871X: nolinked power save enter
[  131.255130] RTL871X: nolinked power save leave
[  133.052601] RTL871X: indicate disassoc
[  139.262696] RTL871X: nolinked power save enter
[  159.745554] RTL871X: nolinked power save leave
[  165.887018] RTL871X: nolinked power save enter
[  192.760404] RTL871X: nolinked power save leave
[  198.908032] RTL871X: nolinked power save enter
[  220.658410] RTL871X: nolinked power save leave
[  226.812445] RTL871X: nolinked power save enter
```

Any hints?

tonsimple, 2017-09-03:
Hey, I got that very behavior when trying to build the driver for Raspbian (as an experiment), thought I just misconfigured something on wpa_supplicant level because wpa_supplicant was complaining about not being able to connect to driver in its logs.

pnpavlov, 2017-09-11:
Yes, seems so... anyone, hint how to solve that one?

guysoft, 2017-12-27:
Same issue here, still no solution?

bermanmaxim, 2018-02-03:
Did you try disabling power save with sudo iw dev <device> set power_state off? I have the same problem, I am trying to see if it helps - for now it does but the problem tends to appear quite randomly (maybe as soon as the power state kicks off).

kimocoder, 2018-02-03:
power saving may be disabled in Makefile too

guysoft, 2019-07-02:
Is there a way to test if it has been disabled? I am still stuck on this.

Akanichi, 2021-06-01:
> Is there a way to test if it has been disabled? I am still stuck on this.

Yes there is this command line to be sure if power save is enabled or not 
$ iw  wlan0 get power_save
wlan0 to replace with the appropriate wifi interface you are using.

But it doesn't resolve the problem to set power_save to off with the command below:
iw  wlan0 set power_save off

So i still get the same message in the log:
**[ 3296.902501] ==> rtl8188e_iol_efuse_patch
[ 3297.206644] R8188EU: nolinked power save leave
[ 3299.211501] R8188EU: nolinked power save enter
**

Jerry-li-0, 2022-02-24:
Does anyone have any other suggestions for the same problem