gnab/rtl8812au #121: Compiling for OpenWRT

tonsimple, 2017-09-03:
Hi! I'm trying to compile it for OpenWRT (have a TP link router, but also a raspberry pi 3 flashed with OpenWRT, so, like, two candidate devices running OpenWRT) and am failing miserably.

I want to use a 8812au based dongle + OpenWRT-device to make an impromptu wifi extender, with the OpenWRT-device's own wifi acting as "access point" and the 8812au device being used as a "client device" and connecting to a very remote access point (the 8812au device has a wicked directional antenna and can reach the remote AP without any issue at all, tested from standalone windows AND debian linux)

I am failing miserably at the "compile for openWRT" part (trying to follow this here https://wiki.openwrt.org/doc/howto/build )

Could someone please help me out? :-)

Chadster766, 2017-09-03:
Why use OpenWRT when Raspbian would work well for this application?

JohnMHarrisJr, 2017-09-06:
@Chadster766 I would assume the reason is the minimal installation size.

linarense, 2017-12-12:
Have you got it @tonsimple ???

EpicLPer, 2018-06-04:
I wonder if you were able to figure something out here

shadow1163, 2019-03-11:
I found a page about openwrt and rlt8812au page in openwrt BBS.
https://forum.openwrt.org/t/rtl8812au-without-drivers/22001.
Here is command:
```
opkg update
opkg install kmod-rtl8812au-ct(openwrt dev build.)
```
My network adapter is D-Link DWA-182, it should work.