gnab/rtl8812au #123: Driver does not automatically reload on restart

chirag7jain, 2017-09-26:
I am using ubuntu 17.10. Installed the drivers using the install.sh script.

Every time I reboot / turn off then then turn on, I have to re run the script.

BTW the script runs fine without any sort of errors. 

Have done the same commands in the script manually earlier and saw no errors.

Alex-Linhares, 2017-10-18:
Have the same issue here (Description:	Linux Mint 18.2 Sonya, 4.8.0-53-generic #56~16.04.1-Ubuntu x86_64), with others [mentioning it in launchpad](https://bugs.launchpad.net/ubuntu/+source/rtl8812au/+bug/1509859).

xdraylin, 2017-12-19:
I would recommend using dkms for installing this driver. Spares you the hassle of executing the installation script every time after a kernel update. You can use install-dkms.sh from my fork of this repository, if you want.

brainthinks, 2017-12-23:
I was having this problem too - I had to unplug the adapter and plug it back in to get internet.  I googled a little and found this:

https://askubuntu.com/questions/794904/usb-wifi-adapter-not-autoloading

Which just so happens to be for this exact driver.  It would be nice to have this line in the readme, assuming this is a common problem:

```
echo 8812au | sudo tee -a /etc/modules
```