gnab/rtl8812au #129: Error during make for raspberry pi

felixgalindo, 2017-12-05:
pi@raspberrypi:~/rtl8812au $ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.9.24-v7+/build M=/home/pi/rtl8812au  modules
make[1]: Entering directory '/lib/modules/4.9.24-v7+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.9.24-v7+/build'
Makefile:1052: recipe for target 'modules' failed
make: *** [modules] Error 2

uname -a
Linux raspberrypi 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux

kimocoder, 2017-12-05:
Do you have the kernel-headers installed?

felixgalindo, 2017-12-05:
Not sure how to install them.

felixgalindo, 2017-12-05:
"sudo apt-get install raspberrypi-kernel-headers" ?

felixgalindo, 2017-12-05:
That solved it thanks.