gnab/rtl8812au #136: Make Error

dazhazard, 2018-02-06:
Any chance you can see the fault?

Ive edited the MAKEFILE as instructed then get the below when running Make.

> root@kali:~/Desktop/rtl8812au-master# make
> 
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.14.0-kali3-amd64/build M=/root/Desktop/rtl8812au-master  modules
> make[1]: Entering directory '/usr/src/linux-headers-4.14.0-kali3-amd64'
>   CC [M]  /root/Desktop/rtl8812au-master/core/rtw_cmd.o
> In file included from /root/Desktop/rtl8812au-master/include/drv_types.h:32:0,
>                  from /root/Desktop/rtl8812au-master/core/rtw_cmd.c:22:
> /root/Desktop/rtl8812au-master/include/osdep_service.h: In function ‘thread_enter’:
> /root/Desktop/rtl8812au-master/include/osdep_service.h:251:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
>   allow_signal(SIGTERM);
>   ^~~~~~~~~~~~
>   do_signal
> /root/Desktop/rtl8812au-master/include/osdep_service.h: In function ‘flush_signals_thread’:
> /root/Desktop/rtl8812au-master/include/osdep_service.h:261:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
>   if (signal_pending (current))
>       ^~~~~~~~~~~~~~
>       timer_pending
> /root/Desktop/rtl8812au-master/include/osdep_service.h:263:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
>    flush_signals(current);
>    ^~~~~~~~~~~~~
>    do_signal
> cc1: some warnings being treated as errors
> /usr/src/linux-headers-4.14.0-kali3-common/scripts/Makefile.build:319: recipe for target '/root/Desktop/rtl8812au-master/core/rtw_cmd.o' failed
> make[4]: *** [/root/Desktop/rtl8812au-master/core/rtw_cmd.o] Error 1
> /usr/src/linux-headers-4.14.0-kali3-common/Makefile:1525: recipe for target '_module_/root/Desktop/rtl8812au-master' failed
> make[3]: *** [_module_/root/Desktop/rtl8812au-master] Error 2
> Makefile:146: recipe for target 'sub-make' failed
> make[2]: *** [sub-make] Error 2
> Makefile:8: recipe for target 'all' failed
> make[1]: *** [all] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-4.14.0-kali3-amd64'
> Makefile:1051: recipe for target 'modules' failed
> make: *** [modules] Error 2
>

NirmithaWithana, 2018-03-17:
I have same issue. Please some one help. I tried a number of things & every time I failed.

etiW, 2019-01-19:
I have the same issue with installing version 3 of TL-WN722N in kali

Juan-Cortez3, 2019-10-22:
Please add the line below in Makefile and try again:
```
EXTRA_CFLAGS += -Wno-implicit-function-declaration
```
I have similar issues because I have a newer version of gcc in my system?
What versions do you have?