gnab/rtl8812au #144: Compilation error on kernel 4.15.0-20-generic (Ubuntu 18.04)

siddharth99, 2018-05-13:
I am running into compilation error with kernel 4.15 since the timer interface has changed.

```
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-20-generic'
  CC [M]  /usr/local/src/rtl8812au/core/rtw_cmd.o
In file included from /usr/local/src/rtl8812au/include/osdep_service.h:41:0,
                 from /usr/local/src/rtl8812au/include/drv_types.h:32,
                 from /usr/local/src/rtl8812au/core/rtw_cmd.c:22:
/usr/local/src/rtl8812au/include/osdep_service_linux.h: In function ‘_init_timer’:
/usr/local/src/rtl8812au/include/osdep_service_linux.h:257:8: error: ‘_timer {aka struct timer_list}’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;
        ^~
/usr/local/src/rtl8812au/include/osdep_service_linux.h:258:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer
```

I made a small change to make it compile but I obviously did something wrong since the code compiles but the kernel hangs as soon as I try to do `insmod 8812au.ko`

```
 __inline static void _init_timer(_timer *ptimer,_nic_hdl nic_hdl,void *pfunc,void* cntx)
 {
        //setup_timer(ptimer, pfunc,(u32)cntx); 
-       ptimer->function = pfunc;
-       ptimer->data = (unsigned long)cntx;
-       init_timer(ptimer);
+       //ptimer->function = pfunc;
+       //ptimer->data = (unsigned long)cntx;
+       // init_timer(ptimer);
+       timer_setup(ptimer, pfunc, (unsigned long)cntx);
 }
```
Any help would be greatly appreciated.

donahue95, 2018-05-14:
https://github.com/gordboy/rtl8812au provides working solution up to 4.17-rc4 
someone with time and interest could borrow and adapt the code

siddharth99, 2018-05-16:
@donahue95 - I tried the link too and I'm able to compile and add the module manually using insmod which seems to work and adds the module. But I don't see the USB wifi device being recognized anywhere.

Jeansen, 2018-05-24:
I also get the same error as @siddharth99 on debian testing with kernel 4.16

kimocoder, 2018-05-24:
You need to manually edit the driver for it to work, as our driver is slighty different, but the patch would work. Take a look at this https://github.com/aircrack-ng/rtl8812au/commit/f221a169f281dab9756a176ec2abd91e0eba7d19

Just manually edit changes

simon0451, 2018-05-28:
@kimocoder There is no hal/phydm/phydem_types.h file in this repo. So I made the changes from the other two modifications from your link  aircrack-ng/rtl8812au@f221a16 and it got a lot further than it was doing before...

But now /home/user/rtl8812au/hal/OUTSRC/odm_interface.c is dropping incompatible pointer type errors. It does not like pTimer...

penguin360, 2018-06-07:
Please someone make it compile and work with Ubuntu 18.04 LTS or the new kernel 4.15 and above?

Please?

bhb27, 2018-06-07:
https://github.com/gnab/rtl8812au/pull/141
resolves this, but doesn't seems to have anyone to merge things here

I just intall using this:
https://github.com/msharov/rtl8812au/commits/master

@gnab 
can you merge the open PR?
Or someone with access please.

penguin360, 2018-06-21:
It's been a while, can someone please do a merge?

kimocoder, 2018-06-21:
@penguin360 then just do "wget https://github.com/msharov/rtl8812au/commit/296fb88cf466fe07fffd1b9d2f42a33c420ceb1d.patch" and patch it yourself? "patch -p1 < *.patch"

penguin360, 2018-06-22:
@kimocoder thanks for your reply. I am just a regular Ubuntu end user and I don't do Linux programming, so I don't know how to do it. I will be very grateful for anyone who can do it.

kimocoder, 2018-06-22:
I just told you.. 

#1 git clone https://github.com/gnab/rtl8812au
#2  cd rtl8812au
#3 wget https://github.com/msharov/rtl8812au/commit/296fb88cf466fe07fffd1b9d2f42a33c420ceb1d.patch

#4 patch -p1 < *.patch
#5 make && make install

Rantelo, 2018-07-04:
@kimocoder solution to patch master worked well for me.
I have Ubuntu 16.04.4 LTS with 4.15.0-24-generic Kernel version.

penguin360, 2018-07-05:
@kimocoder Thank you very much for the patch. It worked like a charm. 

I hope someone can merge it

kimocoder, 2018-07-05:
Your welcome!

mcman56, 2018-07-18:
I'm on 16.04 but get an access warning at the last step.
[image]

mcman56, 2018-07-19:
a slight modification got this to work. For the last command I needed
sudo make && sudo make install
then I needed to reboot ...not that I know why

trentbrown13, 2018-07-19:
Thank you kimocoder, worked for me after changing to sudo make && sudo make install. I did not have to reboot

enrydell, 2018-07-22:
It worked like a charm in Elementary OS kernel 4.15.0-29, I just had to unplug and plug again the adapter.
It's a shame that D-Link only supports til kernel  3.x

Well, thank you all that made it possible!

kimocoder, 2018-07-22:
Think @gnab just merged some pulls, adding kernel v4.14 support and few others.
Takk @gnab - på tide du tok en runde her :+1:

penguin360, 2018-07-25:
@gnab thank you!
@kimocoder does it mean this issue has been fixed and the ticket can be closed?

kimocoder, 2018-07-25:
Exactly. The patch is merged.

mridowijayanto, 2019-10-02:
error what problem here, please halp me

gnumoksha, 2020-01-09:
@mridowijayanto on unbuntu, you can run `apt install rtl8812au-dkms` and its done

hgupta481994, 2020-04-19:
While running command make I am getting this error

hgupta481994, 2020-04-19:
rtl8812au-1/os_dep/linux/os_intfs.c:1415:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
cc1: some warnings being treated as errors

Light--, 2020-10-11:
> I just told you..
> 
> 1: git clone https://github.com/gnab/rtl8812au
> 2: cd rtl8812au
> 3: wget https://github.com/msharov/rtl8812au/commit/296fb88cf466fe07fffd1b9d2f42a33c420ceb1d.patch

the patch is not found!!!

--2020-10-11 09:45:01--  https://github.com/msharov/rtl8812au/commit/296fb88cf466fe07fffd1b9d2f42a33c420ceb1d.patch
Resolving github.com (github.com)... 13.229.188.59
Connecting to github.com (github.com)|13.229.188.59|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-10-11 09:45:02 ERROR 404: Not Found.