gnab/rtl8812au #150: Confirm this driver working for Ubuntu 18.04.1 and TP-Link AC1300 Archer T4U(EU) v2.0

micheldrescher, 2018-08-23:
Kernel version: 4.15.0-32-generic

Would it make sense to start listing confirmed installs with h/w combinations somewhere?

It took me ages and pure luck to find this repository!

micheldrescher, 2018-08-24:
Hi @garceri,

I just checked, and indeed `iw list` doesn't do anything, it just lists nothing. Also, no dropped packages or network instability.

However, `lsusb -v -s 2:2` produces the attached result.

[wifw.txt](https://github.com/gnab/rtl8812au/files/2319693/wifw.txt)

The only curious thing about it is that the driver thinks it is a 802.11n NIC not a 802.11ac NIC - which is no problem for me because I operate only 802.11n networks anyway at the moment.

Is your network a 802.11ac?

jandante, 2019-02-04:
It's not working for the TP-Link AC1300 Archer T4U(EU) v3.0
The interface doesn't get listed;

Slightly better report for https://github.com/abperiasamy/rtl8812AU_8821AU_linux 
That one gives me the adapter but it can not see any networks.

gnumoksha, 2020-01-09:
This driver don't work with Archer T4U v3. [This other driver does work](https://github.com/jeremyb31/rtl8822bu). Also, on ubuntu one can simply execute `apt install rtl8812au-dkms`.