gnab/rtl8812au #153: Slow speed using AWUS036ACH

renno-bih, 2018-10-11:
Hello,

I tried these drivers but speed I'm achieving with AWUS036ACH is not very impressive.
Using iperf and iperf3 I get speeds around **60-80Mbits/sec**. 
Router I'm using is Synology RT2600ac and it is capable of much higher speeds.

By running `iwconfig` I can see that Wireless adapter is operating at 867Mb/s.

> enx00c0caa51893  IEEE 802.11AC  ESSID:"NSoftR&D"  Nickname:"<WIFI@REALTEK>"
>           Mode:Managed  Frequency:5.52 GHz  Access Point: 00:11:32:72:1B:71
>           Bit Rate:867 Mb/s   Sensitivity:0/0
>           Retry:off   RTS thr:off   Fragment thr:off
>           Power Management:off
>           Link Quality=100/100  Signal level=98/100  Noise level=0/100
>           Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
>           Tx excessive retries:0  Invalid misc:0   Missed beacon:0

My laptop has integrated WiFi adapter that is working on 802.11ac standard and I can achieve speeds over 400Mbits/sec. When connecting AWUS036ACH to Windows laptop I also can achieve speeds around 200Mbits/sec.

Any idea what is wrong here?

I'm using **Ubuntu 16.04**, having AWUS036ACH connected to **USB3.0** on **Intel Up Board**.

kimocoder, 2018-10-11:
I may tell you what's wrong.. these drivers from Realtek are really dirty. This issue is related to bad VHT code.

You could try using https://github.com/aircrack-ng/rtl8812au driver instead. Either the v5.2.20 or the (work in progress) version v5.3.4

kimocoder, 2018-10-12:
Make a report over to us, we'll fix it