gnab/rtl8812au #154: driver not working with kernel 4.19 lts

SolZer, 2018-10-26:
upgraded from 4.18 to 4.19, and wifi adapter is no longer recognized

ekeyser, 2018-10-26:
Hi Solzer, did you recompile the driver? When kernel upgrades occur the driver modules are not copied over to the new kernel. Usually a `make; sudo make install` (and then reboot or `insmod`) should correct this. Just checking.

kimocoder, 2018-10-26:
Kernel v4.19 will need a patch to her supported. Take a look in "gordboy" repo

jasontm7, 2018-11-21:
Hi ekeyser,
I know this thread is getting a little old but I had the same issue when Fedora upgraded to 4.19 kernel.  My particular issue was a duplicate #define error in ./include/wifi.h: Line 980: #define IEEE80211_MAX_AMPDU_BUF 0x40 which was also defined elsewhere as #define IEEE80211_MAX_AMPDU_BUF 0x100.
So I commented out this #define in wifi.h and let it compile: /*#define IEEE80211_MAX_AMPDU_BUF 0x40*/
This worked for me.  Can't guarantee it'll work for everyone but it's something to try if nothing else works and it's easy. I know there's probably a reason to make this particular buffer smaller but it maxes out my internet speed and works after reboots so I don't care :-)

romulasry, 2018-12-05:
https://github.com/gordboy/rtl8812au

fffaraz, 2018-12-21:
I guess I have the same problem.

`
 CC [M]  /home/faraz/rtl8812au-master/os_dep/linux/os_intfs.o
/home/faraz/rtl8812au-master/os_dep/linux/os_intfs.c:1069:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/home/faraz/rtl8812au-master/os_dep/linux/os_intfs.c:1069:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
scripts/Makefile.build:303: recipe for target '/home/faraz/rtl8812au-master/os_dep/linux/os_intfs.o' failed
make[2]: *** [/home/faraz/rtl8812au-master/os_dep/linux/os_intfs.o] Error 1
Makefile:1517: recipe for target '_module_/home/faraz/rtl8812au-master' failed
make[1]: *** [_module_/home/faraz/rtl8812au-master] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.10-041910-generic'
Makefile:1064: recipe for target 'modules' failed
make: *** [modules] Error 2
`
`
gcc --version
gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
`
`
uname -r -v -m
4.19.10-041910-generic #201812170433 SMP Mon Dec 17 09:35:34 UTC 2018 x86_64
`

mirix, 2019-01-08:
Compare Makefile and os_dep/linux/os_intfs.c between gnab and gordboy versions.

In the Makefile there are a few extra flags that differ. My guess is that this is irrelevant but I changed them anyway so that both files match.

In os_dep/linux/os_intfs.c search for rtw_select_queue. There are three occurrences. The last two are identical at the time of this writing. The first is within a series of if statements. The gordboy version has an extra one for kernel 4.19.

Replace the gnab chunk with the gordboy version and you are done: it compiles and works.

mishaturnbull, 2019-01-24:
Same problem now --

Kali Linux, uname -a -> `Linux [redacted] 4.19.0-kali1-amd64 #1 SMP Debian 4.19.13-1kali1 (2019-01-03) x86_64 GNU/Linux`

Exact same error as @chovy when trying to `make` the driver.  It may be worth noting that the guide I followed suggested an `apt install linux-headers-generic` before even cloning this repo, and on Kali, I cannot find that package.

mirix, 2019-01-24:
Yesterday I did a system upgrade and Liquorix 4.20 was installed. The module was build automatically via DKMS. This means that the simple hack described above works for versions 4.19 and 4.20.

mishaturnbull, 2019-01-24:
I've had no luck installing via DKMS or the gordboy repo, but what I was able to do was use the CD drive that came with the dongle to install the Bluetooth driver, and the WiFi driver via https://github.com/lwfinger/8723bu.  If anyone's interested I can post the code from the CD drive.

chovy, 2019-01-24:
I'm sorry I misled you. I thought this was about sound drivers for my laptop, not the wifi card. I don't know how to get it to work.

mirix, 2019-01-24:
Misha, the gordboy code doesn't work for me either. What works is hacking a couple of files from this repo as explained above.

mirix, 2019-02-01:
No need to touch the Makefile. Just replace os_dep/linux/os_intfs.c with the attached file (change extension from .txt to .c) and your are good to go.

[os_intfs.txt](https://github.com/gnab/rtl8812au/files/2822288/os_intfs.txt)

mirix, 2019-02-05:
Hi guys, the aircrack-ng/rtl8812au drivers work for me. I have tested v5.3.4 of the driver against the 4.19.19 kernel. It works out of the box (DKMS method), the repo seems to be actively maintained, it wraps all the commands in convenient install/uninstall scripts, and it has contains more recent versions of the drivers (not the most recent, though).

chovy, 2019-02-05:
@mirix where can i read more about this?

mirix, 2019-02-06:
@chovy 

https://github.com/aircrack-ng/rtl8812au

AstroDrabb, 2019-03-24:
> upgraded from 4.18 to 4.19, and wifi adapter is no longer recognized

I have made a fix so it works with 4.19.* and also 5.0.* here:
https://github.com/AstroDrabb/rtl8812au

chovy, 2019-03-24:
Can you give me foil proof instructions for installing the driver in Ubuntu?

-
Anthony

AstroDrabb, 2019-03-26:
It has been a long time since I used Ubuntu, however the steps to try are:

git clone https://github.com/AstroDrabb/rtl8812au
sudo mkdir -p /usr/src # might already exist
sudo cp -r rtl8812au /usr/src

sudo dkms add -m rtl8812au -v 4.2.2 --all
sudo dkms build -m rtl8812au -v 4.2.2 -k $(uname -r)
sudo dkms install -m rtl8812au -v 4.2.2 -k $(uname -r)

the driver should now be installed.  Network Manager should give you a
notice.
If not, reboot and go in to Network Manager and look for you USB adapter and
configure it.

Let me know if you need any more help.

Jim

hoemon, 2019-06-08:
step by step instructions for manjaro is also very much appreciated..
I can't seem to make this work :(

NTOTL, 2019-07-16:
> It has been a long time since I used Ubuntu, however the steps to try are: git clone https://github.com/AstroDrabb/rtl8812au sudo mkdir -p /usr/src # might already exist sudo cp -r rtl8812au [...]
I am running on Kernel 5.0 and the install failed at the last step:

8812au:
Running module version sanity check.
/usr/sbin/dkms: line 784: find_module: command not found
modinfo: ERROR: missing module or filename.
 - Original module
 - Installation
   - Installing to /lib/modules/5.0.0-21-generic/updates/dkms/
Can someone help?

knassar702, 2019-10-02:
help me please i cant install realtek 8188 

The Error 

````
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-4.19.0-kali4-common/scripts/Makefile.build:309: /root/linux/RTL8188EUS_linux_v4.1.4_6773.20130222/driver/rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_cmd.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.19.0-kali4-common/Makefile:1535: _module_/root/linux/RTL8188EUS_linux_v4.1.4_6773.20130222/driver/rtl8188EUS_linux_v4.1.4_6773.20130222] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-kali4-686-pae'
make: *** [Makefile:678: modules] Error 2
##################################################
Compile make driver error: 2
Please check error Mesg
##################################################

````

My OS : Kali Linux  2019 (32)

donahue95, 2019-10-02:
RTL8188EUS is not in the 8812AU family. You need a driver made for your chipset. If you google for ' github RTL8188EUS ' you will bring up several repos for 8188EUS drivers.