gnab/rtl8812au #165: [kernel 5.0>] compile error

yamal4321, 2019-07-31:
when running make the following error appears:
**error: macro "access_ok" passed 3 arguments, but takes just 2
error: ‘access_ok’ undeclared (first use in this function)
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)){
**

fixed by:
-access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)
+access_ok(priv_cmd.buf, priv_cmd.total_len)

gnumoksha, 2020-01-08:
I think this error was fixed in the last releases.

error10, 2021-02-15:
Still happening in master today.

```
/var/lib/dkms/8812au/4.2.3/build/os_dep/linux/rtw_android.c: In function ‘rtw_android_priv_cmd’:
/var/lib/dkms/8812au/4.2.3/build/os_dep/linux/rtw_android.c:356:63: error: macro "access_ok" passed 3 arguments, but takes just 2
   if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
                                                               ^
/var/lib/dkms/8812au/4.2.3/build/os_dep/linux/rtw_android.c:356:8: error: ‘access_ok’ undeclared (first use in this function)
   if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
        ^~~~~~~~~
/var/lib/dkms/8812au/4.2.3/build/os_dep/linux/rtw_android.c:356:8: note: each undeclared identifier is reported only once for each function it appears in
```

This on 4.18.0-240.10.1.el8_3.x86_64