gnab/rtl8812au #170: Make error

mbwmbw1337, 2019-09-03:
Running Kali Linux on Raspberry Pi

Linux raspberry-pi-2 4.19.66-Re4son-v8+ #1 SMP PREEMPT Sun Aug 18 12:14:35 UTC 2019 aarch64 GNU/Linux

`root@raspberry-pi-2:/usr/src/8812au-4.2.2# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.66-Re4son-v8+/build M=/usr/src/8812au-4.2.2  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.66-Re4son-v8+'
  CC [M]  /usr/src/8812au-4.2.2/core/rtw_cmd.o
gcc: error: unrecognized argument in option ‘-mabi=apcs-gnu’
gcc: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
gcc: error: unrecognized command line option ‘-mapcs’
gcc: error: unrecognized command line option ‘-mno-sched-prolog’
gcc: error: unrecognized command line option ‘-msoft-float’
make[2]: *** [scripts/Makefile.build:304: /usr/src/8812au-4.2.2/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1519: _module_/usr/src/8812au-4.2.2] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.66-Re4son-v8+'
make: *** [Makefile:1065: modules] Error 2
`

99Kies, 2019-09-24:
running ubuntu16.04 . . . too，have skill ?

geerlingguy, 2020-11-09:
Related: https://github.com/gnab/rtl8812au/issues/25#issuecomment-527273047