gnab/rtl8812au #173: RPi 3B, Buster (4.19.75-v7+): Exec format error when Make

Gooseman42, 2019-10-11:
after installing on a fresh buster distribution, I get the following error while making:

`make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.75-v7+/build M=/home/pi/rtl8812au  modules`
`make[1]: Entering directory '/usr/src/linux-headers-4.19.75-v7+'`
`  CC [M]  /home/pi/rtl8812au/core/rtw_cmd.o`
`/bin/sh: 1: scripts/basic/fixdep: Exec format error`
`make[2]: *** [scripts/Makefile.build:304: /home/pi/rtl8812au/core/rtw_cmd.o] Error 2`
`make[1]: *** [Makefile:1519: _module_/home/pi/rtl8812au] Error 2`
`make[1]: Leaving directory '/usr/src/linux-headers-4.19.75-v7+'`
`make: *** [Makefile:1064: modules] Error 2`

Does 'Exec format error' point to a problem with the kernel headers?

twojciac, 2019-10-11:
I'm running into the same issues, but found this is related to another bug:

https://github.com/raspberrypi/linux/issues/3279

Gooseman42, 2019-10-14:
I can confirm the issue has been fixed in the newest distribution containing:
raspberrypi-kernel                         1.20190925+1-1
raspberrypi-kernel-headers           1.20190925+1-1

closed 2019-10-14 by Gooseman42