gnab/rtl8812au #177: Compilation error on kernel 5.4

danwerspb, 2019-12-11:
Environment: Manjaro Linux, kernel 5.4

Message when i compile driver:

```
user@sev-lucky  /usr/src/8812au-4.2.2 ● master ●  sudo -E ./install.sh
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.2-1-MANJARO/build M=/usr/src/8812au-4.2.2  modules
make[1]: вход в каталог «/usr/lib/modules/5.4.2-1-MANJARO/build»
  CC [M]  /usr/src/8812au-4.2.2/core/rtw_mlme_ext.o
/usr/src/8812au-4.2.2/core/rtw_mlme_ext.c: В функции «mgt_dispatcher»:
/usr/src/8812au-4.2.2/core/rtw_mlme_ext.c:712:20: ошибка: expected «)» before «__attribute__»
  712 |    __attribute__ ((fallthrough));
      |                    ^
      |                    )
/usr/src/8812au-4.2.2/core/rtw_mlme_ext.c:712:32: ошибка: expected identifier or «(» before «)» token
  712 |    __attribute__ ((fallthrough));
      |                                ^
/usr/src/8812au-4.2.2/core/rtw_mlme_ext.c:712:4: предупреждение: ISO C90 запрещает смешение деклараций и кода [-Wdeclaration-after-statement]
  712 |    __attribute__ ((fallthrough));
      |    ^~~~~~~~~~~~~
/usr/src/8812au-4.2.2/core/rtw_mlme_ext.c:708:6: предупреждение: этот оператор может провалиться [-Wimplicit-fallthrough=]
  708 |    if(check_fwstate(pmlmepriv, WIFI_AP_STATE) == _TRUE)
      |      ^
/usr/src/8812au-4.2.2/core/rtw_mlme_ext.c:713:3: замечание: здесь
  713 |   case WIFI_ASSOCREQ:
      |   ^~~~
make[2]: *** [scripts/Makefile.build:266: /usr/src/8812au-4.2.2/core/rtw_mlme_ext.o] Ошибка 1
make[1]: *** [Makefile:1652: /usr/src/8812au-4.2.2] Ошибка 2
make[1]: выход из каталога «/usr/lib/modules/5.4.2-1-MANJARO/build»
make: *** [Makefile:1065: modules] Ошибка 2

```

There i no problem nn kernel 5.3 (!)

tobiichiamane, 2020-01-04:
Please try to change line 712 in file `core/rtw_mlme_ext.c` from `__attribute__ ((fallthrough));` to `__attribute__ ((__fallthrough__));` and try again.

johnperry-math, 2020-01-07:
This works for me on Fedora 30. Any chance of getting it into the main distribution?

tobiichiamane, 2020-01-11:
> This works for me on Fedora 30. Any chance of getting it into the main distribution?

It may not be a good idea, because it may break the project in other environments.

closed 2020-02-13 by dpvpro