gnab/rtl8812au #178: Compilation Error for Kernel 5.4.2 - Ubuntu 18.04.3 LTS

wjsilva85, 2019-12-12:
asm_inline volatile("1:\t" ins "\n"    \
  ^~~~~~~~~~
./arch/x86/include/asm/bug.h:79:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, BUGFLAG_WARNING|(flags));  \
  ^~~~~~~~~~
./include/asm-generic/bug.h:90:19: note: in expansion of macro ‘__WARN_FLAGS’
 #define __WARN()  __WARN_FLAGS(BUGFLAG_TAINT(TAINT_WARN))
                   ^~~~~~~~~~~~
./include/asm-generic/bug.h:115:3: note: in expansion of macro ‘__WARN’
   __WARN();      \
   ^~~~~~
./include/linux/usb.h:1969:3: note: in expansion of macro ‘WARN_ON’
   WARN_ON(usb_pipeout(pipe));
   ^~~~~~~
scripts/Makefile.build:265: recipe for target '/home/wellington/rtl8812au/core/rtw_cmd.o' failed
make[2]: *** [/home/wellington/rtl8812au/core/rtw_cmd.o] Error 1
Makefile:1652: recipe for target '/home/wellington/rtl8812au' failed
make[1]: *** [/home/wellington/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.2-050402-generic'
Makefile:1065: recipe for target 'modules' failed
make: *** [modules] Error 2
wellington@well-personal-pc:~/rtl8812au$ clear

wellington@well-personal-pc:~/rtl8812au$ ls
8812au.ko     8812au.o   hal          Kconfig         os_dep
8812au.mod    clean      ifcfg-wlan0  Makefile        README.md
8812au.mod.c  core       include      modules.order   runwpa
8812au.mod.o  dkms.conf  install.sh   Module.symvers  wlan0dhcp
wellington@well-personal-pc:~/rtl8812au$ sudo make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.2-050402-generic/build M=/home/wellington/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.2-050402-generic'
  CC [M]  /home/wellington/rtl8812au/core/rtw_cmd.o
In file included from <command-line>:0:0:
./arch/x86/include/asm/page_64.h: In function ‘clear_page’:
././include/linux/compiler_types.h:210:24: error: expected ‘(’ before ‘__inline’
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/alternative.h:256:2: note: in expansion of macro ‘asm_inline’
  asm_inline volatile (ALTERNATIVE_2("call %P[old]", "call %P[new1]", feature1,\
  ^~~~~~~~~~
./arch/x86/include/asm/page_64.h:49:2: note: in expansion of macro ‘alternative_call_2’
  alternative_call_2(clear_page_orig,
  ^~~~~~~~~~~~~~~~~~
In file included from ./arch/x86/include/asm/barrier.h:5:0,
                 from ./include/linux/compiler.h:256,
                 from ./include/linux/export.h:42,
                 from ./include/linux/linkage.h:7,
                 from ./include/linux/preempt.h:10,
                 from ./include/linux/spinlock.h:51,
                 from /home/wellington/rtl8812au/include/osdep_service_linux.h:24,
                 from /home/wellington/rtl8812au/include/osdep_service.h:41,
                 from /home/wellington/rtl8812au/include/drv_types.h:32,
                 from /home/wellington/rtl8812au/core/rtw_cmd.c:22:
./arch/x86/include/asm/alternative.h:128:2: error: expected identifier or ‘(’ before string constant
  "# ALT: oldinstr2\n"         \
  ^
./arch/x86/include/asm/alternative.h:168:2: note: in expansion of macro ‘OLDINSTR_2’
  OLDINSTR_2(oldinstr, 1, 2)     \
  ^~~~~~~~~~
./arch/x86/include/asm/alternative.h:256:23: note: in expansion of macro ‘ALTERNATIVE_2’
  asm_inline volatile (ALTERNATIVE_2("call %P[old]", "call %P[new1]", feature1,\
                       ^~~~~~~~~~~~~
./arch/x86/include/asm/page_64.h:49:2: note: in expansion of macro ‘alternative_call_2’
  alternative_call_2(clear_page_orig,
  ^~~~~~~~~~~~~~~~~~
In file included from <command-line>:0:0:
././include/linux/compiler_types.h:210:24: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/alternative.h:256:2: note: in expansion of macro ‘asm_inline’
[... 3709 lines cut ...]
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/spinlock.h:51,
                 from /home/wellington/rtl8812au/include/osdep_service_linux.h:24,
                 from /home/wellington/rtl8812au/include/osdep_service.h:41,
                 from /home/wellington/rtl8812au/include/drv_types.h:32,
                 from /home/wellington/rtl8812au/core/rtw_cmd.c:22:
./arch/x86/include/asm/bug.h:35:22: error: expected identifier or ‘(’ before string constant
  asm_inline volatile("1:\t" ins "\n"    \
                      ^
./arch/x86/include/asm/bug.h:79:2: note: in expansion of macro ‘_BUG_FLAGS’
  _BUG_FLAGS(ASM_UD2, BUGFLAG_WARNING|(flags));  \
  ^~~~~~~~~~
./include/asm-generic/bug.h:90:19: note: in expansion of macro ‘__WARN_FLAGS’
 #define __WARN()  __WARN_FLAGS(BUGFLAG_TAINT(TAINT_WARN))
                   ^~~~~~~~~~~~
./include/asm-generic/bug.h:115:3: note: in expansion of macro ‘__WARN’
   __WARN();      \
   ^~~~~~
./include/linux/sched/signal.h:482:3: note: in expansion of macro ‘WARN_ON’
   WARN_ON(!test_thread_flag(TIF_SIGPENDING));
   ^~~~~~~
In file included from <command-line>:0:0:
././include/linux/compiler_types.h:210:24: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 #define asm_inline asm __

Jmt1995, 2022-10-12:
the same problem

Yanbing-Shao, 2023-09-12:
same problem

ljf10000, 2024-09-27:
same problem