gnab/rtl8812au #188: support ubuntu v20.04 ? kernel 5.4+

station19, 2020-04-08:
support ubuntu v20.04 ? kernel 5.4+

nickm324, 2020-04-26:
I need this also, any plans to update it?

warezit, 2020-04-26:
I have this working in Ubuntu 20.04 using Asus USB AC56 adapter ... 5G wifi doesn't seem to work very well, but 2.4G seems to be stable

GerardoBaeza, 2020-05-01:
i have problems installing this driver for the tplink archer t2u (2357:011f TP-Link 802.11ac WLAN Adapter) is someone able to make it work already?

GerardoBaeza, 2020-05-01:
DKMS make.log for rtl8812au-5.3.4 for kernel 5.4.0-28-generic (x86_64)
vie 01 may 2020 15:07:48 CDT
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-28-generic/build M=/var/lib/dkms/rtl8812au/5.3.4/build O="" modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_com.o
/var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.c:1415:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
 1415 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.c:1415:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_com_phycfg.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_btcoex_wifionly.o
cc1: some warnings being treated as errors
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/hal/hal_btcoex.o
make[2]: *** [scripts/Makefile.build:275: /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_cfg80211.c:9512:25: warning: initialized field overwritten [-Woverride-init]
 9512 |  .set_monitor_channel = cfg80211_rtw_set_monitor_channel,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/ioctl_cfg80211.c:9512:25: note: (near initialization for ‘rtw_cfg80211_ops.set_monitor_channel’)
make[1]: *** [Makefile:1719: /var/lib/dkms/rtl8812au/5.3.4/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'
make: *** [Makefile:1711: modules] Error 2

GerardoBaeza, 2020-05-01:
To everyone failing to compile this driver, go to https://github.com/gordboy/rtl8812au-5.6.4.2