gnab/rtl8812au #193: Device added as Ethernet interface instead of wireless interface

tomty89, 2020-06-09:
Hence completely unusable:
```
[tom@archlinux ~]$ ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: enp3s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master bridge0 state UP mode DEFAULT group default qlen 1000
    link/ether ba:f0:20:42:64:5f brd ff:ff:ff:ff:ff:ff permaddr ac:22:0b:29:e6:0c
3: bridge0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether ac:22:0b:29:e6:0c brd ff:ff:ff:ff:ff:ff
[tom@archlinux ~]$ sudo modprobe 8812au
[tom@archlinux ~]$ ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: enp3s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master bridge0 state UP mode DEFAULT group default qlen 1000
    link/ether ba:f0:20:42:64:5f brd ff:ff:ff:ff:ff:ff permaddr ac:22:0b:29:e6:0c
3: bridge0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether ac:22:0b:29:e6:0c brd ff:ff:ff:ff:ff:ff
13: enp0s20u2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether cc:e1:d5:29:fd:58 brd ff:ff:ff:ff:ff:ff
[tom@archlinux ~]$ iw list
[tom@archlinux ~]$ 
```

tomty89, 2020-06-09:
I'm using 0f92542 with 5.6.15 kernel on Arch Linux btw

alanfanh, 2020-09-02:
hi, I have the same problem with you. Do you solved this ?

tomty89, 2020-09-02:
@alanfanh Use this instead: https://github.com/aircrack-ng/rtl8812au

closed 2020-09-03 by tomty89