gnab/rtl8812au #195: Compilation breaks with kernel 5.8.0-rc1

githubris, 2020-06-16:
Hi, thanks for the great work! The driver works fine with the latest stable 5.7.2 kernel but breaks with latest mainline 5.8.0-rc1. Much appreciated if you could look into this new kernel series.

$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-rc1/build M=/SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master  modules
make[1]: Entering directory '/root/KERNELS/linux-5.8-rc1'
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_cmd.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_security.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_debug.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_io.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_ioctl_query.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_ioctl_set.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_ieee80211.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_mlme.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_mlme_ext.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_wlan_util.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_vht.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_pwrctrl.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_rf.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_recv.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_sta_mgt.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_ap.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_xmit.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_p2p.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_tdls.o
  CC [M]  /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_br_ext.o
In file included from /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/include/drv_types.h:80,
                 from /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_br_ext.c:32:
/SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/include/rtw_security.h:194:8: error: redefinition of ‘struct sha256_state’
  194 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/net/inet_sock.h:22,
                 from ./include/net/ip.h:27,
                 from /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_br_ext.c:24:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:281: /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master/core/rtw_br_ext.o] Error 1
make[1]: *** [Makefile:1764: /SEAGATE_03/RTL8812AU-REALTEK-WIFI/rtl8812au-master] Error 2
make[1]: Leaving directory '/root/KERNELS/linux-5.8-rc1'
make: *** [Makefile:1068: modules] Error 2

githubris, 2020-06-16:
There's a patch for another rtl8812au driver which also broke with kernel 5.8.0-rc1 at https://github.com/aircrack-ng/rtl8812au/issues/658. I think this might provide some clues about new structure introduced by this kernel.

dfriede, 2020-08-11:
### Name clash in 5.8 struct sha256_state
renamed to rtw_sha256_state
[diff.txt](https://github.com/gnab/rtl8812au/files/5057679/diff.txt)

DiabeticCrab, 2020-08-14:
Here is a properly formatted patch. I adapted the changes from the aircrack-ng repo for this:
[0001-8812au-linux5.8.patch](https://github.com/gnab/rtl8812au/files/5072086/0001-8812au-linux5.8.patch.txt)