gnab/rtl8812au #198: Driver fails to work with new kernel (Ubuntu 20.04)

adefaria, 2020-06-25:
Hello,

I seem to have hit a problem much like #182 in that the driver fails on the latest kernel I've installed. Works on previous kernel (and for several kernels now) bu not on the latest:

Works with 5.4.0-37-generic
Fails with 5.4.0-39-generic

I also would like to see this either integrated into the Ubuntu repos or a PPA made.

Let me know if you need any more information and how I can provide that.

Thanks.

netqyq, 2020-07-05:
@adefaria Have you solved this? I have the same problem, kernel is 5.4.0-40-generic.

adefaria, 2020-08-28:
Yes, I solved this by buying a brand new system with a good Wifi from System76. One of the biggest reasons to ditch my 15-year-old desktop which served me very well was the lack of support on the older hardware and this wifi issue was the straw that broke the camel's back for me.

On the plus side, I now have a honking new desktop with tons-o-memory and CPU, which is totally Linux supported hardware wise and should last me the rest of my life.

closed 2020-08-28 by adefaria