gnab/rtl8812au #219: build failed on kernel version 5.4.0-58-generic

mbs0221, 2021-01-04:
ubuntu@optiplex-5060:~$ sudo dkms build 8812au/4.2.2

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.4.0-58-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for 8812au: 4.2.2 not found
Error! Bad return status for module build on kernel: 5.4.0-58-generic (x86_64)
Consult /var/lib/dkms/8812au/4.2.2/build/make.log for more information.

pakair, 2021-01-16:
do you have the headers for that kernel (linux-headers-5.4.0-58-generic) ?

$ git clone https://github.com/gnab/rtl8812au
$ cd rtl8812au && sudo dkms add ./rtl8812au
$ sudo dkms install 8812au/4.2.2

TCROC, 2021-01-23:
I get a similar error on Ubuntu 18.04 with:

```
Building module:
cleaning build area...(bad exit status: 2)
...(bad exit status: 2)4-generic
ERROR (dkms apport): binary package for 8812au: 4.2.3 not found
Error! Bad return status for module build on kernel: 5.4.0-64-generic (x86_64)
Consult /var/lib/dkms/8812au/4.2.3/build/make.log for more information.
Makefile:1097: recipe for target 'dkms_install' failed
make: *** [dkms_install] Error 10
```

and makefile log

```
DKMS make.log for 8812au-4.2.3 for kernel 5.4.0-64-generic (x86_64)
Sat Jan 23 12:10:02 EST 2021
make[1]: Entering directory '/var/lib/dkms/8812au/4.2.3/build'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-64-generic
/build M=/var/lib/dkms/8812au/4.2.3/build  modules
make[2]: Entering directory '/var/lib/dkms/8812au/4.2.3/build'
make[2]: *** /lib/modules/5.4.0-64-generic
/build: No such file or directory.  Stop.
make[2]: Leaving directory '/var/lib/dkms/8812au/4.2.3/build'
Makefile:1077: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8812au/4.2.3/build'
```

makangzhe, 2021-11-09:
have you  solve it ?

pakair, 2021-11-09:
Is the question for me? I just upgraded my kernel to 5.4.0-90 It rebuilt it for 4.5.0-90

```
sudo dkms status
8812au, 4.2.2, 5.4.0-88-generic, x86_64: installed
8812au, 4.2.2, 5.4.0-89-generic, x86_64: installed
8812au, 4.2.2, 5.4.0-90-generic, x86_64: installed
```