gnab/rtl8812au #240: Will not compile because of net/ipx.h

mtdew365, 2021-12-27:
Fedora 34. Kernel 4.15.11-100.

I have kernel-core, kernel-devel, kernel-headers, kernel-modules, kernel-modules-extra all loaded.

Received the following error when compiling source using provided install.sh program...
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.11-100.fc34.x86_64/build M=/home/jbergmann/rtl8812au  modules
make[1]: Entering directory '/usr/src/kernels/5.15.11-100.fc34.x86_64'
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_cmd.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_security.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_debug.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_io.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_mlme.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_vht.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_rf.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_recv.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_ap.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_xmit.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_p2p.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_tdls.o
  CC [M]  /home/jbergmann/rtl8812au/core/rtw_br_ext.o
/home/jbergmann/rtl8812au/core/rtw_br_ext.c:25:10: fatal error: net/ipx.h: No such file or directory
   25 | #include <net/ipx.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:277: /home/jbergmann/rtl8812au/core/rtw_br_ext.o] Error 1
make[1]: *** [Makefile:1872: /home/jbergmann/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.15.11-100.fc34.x86_64'
make: *** [Makefile:1086: modules] Error 2

IPX is not supported by the linux kernel since v5.15-rc1 see [torvalds/linux@6c9b408](https://github.com/torvalds/linux/commit/6c9b40844751ea30c72f7a2f92f4d704bc6b2927)

Thanks.

StyraxBenzoin, 2022-01-01:
Nothing to add, but I've got the same issue. Kernel: Linux 5.15.8-76051508-generic

jodydadescott, 2022-02-05:
I forked and removed the IPX code
https://github.com/jodydadescott/rtl8812au

LorenzoAncora, 2022-02-06:
@jodydadescott it seems @fsantini added a retrocompatible version of your changes in https://github.com/gnab/rtl8812au/pull/237. Maybe a fork is superfluous.