gnab/rtl8812au #242: Cannot compile driver

The-Epic-Emerald, 2022-03-19:
[image]

L-m-b, 2022-04-17:
i have this issue too

The-Epic-Emerald, 2022-04-18:
Try using this instead, works fine on latest Kernel:
https://github.com/aircrack-ng/rtl8812au

L-m-b, 2022-04-19:
> Try using this instead, works fine on latest Kernel: https://github.com/aircrack-ng/rtl8812au
> […](#)

I found the best driver to use and that doesn't have this issue is a repo called rtl88 it has patched in the AU driver as well as the PCIE variant I NEEDED which is RTL8812CE or RTL8812C. THAT us the best one to use for Realtek wifi cards including the TPLINK Ac600 PCIE which I have.