gnab/rtl8812au #248: kthread_complete_and_exit undefined (Ubuntu 22.04 latest update)

finleyhust, 2022-07-02:
make the project under  Ubuntu 22.04, latest update.

/home/ubuntu/download/rtl8812au-master/core/rtw_cmd.c:557:1: error: control reaches end of non-void function [-Werror=return-type]
  557 | }
      | ^
cc1: some warnings being treated as errors

/home/ubuntu/download/rtl8812au-master/core/rtw_cmd.c:557:1: error: control reaches end of non-void function [-Werror=return-type]
  557 | }
      | ^
cc1: some warnings being treated as errors

and finally:

entering “/usr/src/linux-headers-5.15.0-40-generic”
  CC [M]  /home/ubuntu/download/rtl8812au-master/core/rtw_mp.o
  CC [M]  /home/ubuntu/download/rtl8812au-master/core/rtw_mp_ioctl.o
  LD [M]  /home/ubuntu/download/rtl8812au-master/8812au.o
  MODPOST /home/ubuntu/download/rtl8812au-master/Module.symvers
ERROR: modpost: "kthread_complete_and_exit" [/home/ubuntu/download/rtl8812au-master/8812au.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:134：/home/ubuntu/download/rtl8812au-master/Module.symvers] 
..................

CharlesButcher, 2022-07-02:
I believe I'm seeing the first of these errors when I try to compile for Raspberry Pi (Debian 5.10). I'm by no means a programmer and am following the simple walkthrough [here](https://alexatnet.com/using-rtl8812-wi-fi-adapter-with-raspberry-pi). I think I've edited the makefile correctly (`CONFIG_PLATFORM_I386_PC = n`, `CONFIG_PLATFORM_ARM_RPI = y`) and the kernel headers are installed.

```
pi@hedgehogcam:~/rtl8812au $ sudo ./install.sh
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.10.103+/build M=/home/pi/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.103+'
  CC [M]  /home/pi/rtl8812au/core/rtw_cmd.o
/home/pi/rtl8812au/core/rtw_cmd.c: In function ‘rtw_cmd_thread’:
/home/pi/rtl8812au/core/rtw_cmd.c:557:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:280: /home/pi/rtl8812au/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1825: /home/pi/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.103+'
make: *** [Makefile:1086: modules] Error 2
```

heng-0-mat, 2022-07-23:
Had the same error and landed here. Came up with a **quick fix** which made it possible to compile and use on rbpi2 running:
Linux raspberrypi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux

To get rid of compiler complaints about missing return statements I added in rtw_cmd.c 555 (after thread_exit() call):
`return ret;`

I assume since thread_exit() will exit the thread the return statement should never be reached but the compiler does not know about that.
Similar error will occur in rtw_mp.c and rtw_xmit.c after calls to thread_exit() and can be addressed in the same way. I copied the declaration of ret from the function definition in rtw_cmd.c i.e.
`u8 ret;`
...
`return ret;`

Also important depending on your kernel version is the renaming of "kthread_complete_and_exit" which can be done in include/osdep_service_linux.h (explained here: https://github.com/gnab/rtl8812au/issues/250 and there is already a pull request  https://github.com/gnab/rtl8812au/pull/246)