gnab/rtl8812au #250: Line 557 in rtl8812au/core/rtw_cmd.c i get the following error: control reaches end of non-void function

wxqg, 2022-07-06:
When I run make to build the driver I get the following errors.

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.13.0-52-generic/build M=/home/nicholas/wifidriver2/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-52-generic'
  CC [M]  /home/nicholas/wifidriver2/rtl8812au/core/rtw_cmd.o
/home/nicholas/wifidriver2/rtl8812au/core/rtw_cmd.c: In function ‘rtw_cmd_thread’:
/home/nicholas/wifidriver2/rtl8812au/core/rtw_cmd.c:557:1: error: control reaches end of non-void function [-Werror=return-type]
  557 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /home/nicholas/wifidriver2/rtl8812au/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1879: /home/nicholas/wifidriver2/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-52-generic'
make: *** [Makefile:1086: modules] Error 2

magicly, 2022-07-07:
原来是kernel 5.17改名字了， 我的kernel是5.13的。 grep一下kthread_complete_and_exit， 发现是在include/osdep_service_linux.h 149行， 改为如下即可：
```code
#define thread_exit() complete_and_exit(NULL, 0)
```

参考 Kernel 5.17 renames complete_and_exit to kthread_complete_and_exit https://github.com/lwfinger/rtl8723ds/issues/15
https://lore.kernel.org/lkml/20211208202532.16409-8-ebiederm@xmission.com/

magicly, 2022-07-07:
测试发现， 其实第一个错误可以不管， 只需要改`thread_exit`定义即可：

```code
diff --git a/include/osdep_service_linux.h b/include/osdep_service_linux.h
index 32a55b9..44fa5d5 100644
--- a/include/osdep_service_linux.h
+++ b/include/osdep_service_linux.h
@@ -146,7 +146,7 @@
        typedef int             thread_return;
        typedef void*   thread_context;

-       #define thread_exit() kthread_complete_and_exit(NULL, 0)
+       #define thread_exit() complete_and_exit(NULL, 0)^M

        typedef void timer_hdl_return;
        typedef void* timer_hdl_context;

```

wxqg, 2022-07-08:
@magicly What you told me works. Thanks a lot magicly!

magicly, 2022-07-08:
My pleasure~

topherbuckley, 2022-07-28:
This is address in [this PR](https://github.com/gnab/rtl8812au/pull/249)