gnab/rtl8812au #252: "Skipping BTF generation for /home/kali/Downloads/rtl8812au/8812au.ko due to unavailability of vmlinux"

araujo88, 2022-08-26:
I installed this module on 5.18.0-kali5-amd64 using `./install.sh` but it doesn't load (only the wired adapter shows up). Here is my installation log:

```
$ sudo ./install.sh                
[sudo] password for kali: 
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.0-kali5-amd64/build M=/home/kali/Downloads/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.18.0-kali5-amd64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.3.0-3) 11.3.0
  You are using:           gcc-11 (Debian 11.3.0-5) 11.3.0
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_cmd.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_security.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_debug.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_io.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_mlme.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_vht.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_rf.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_recv.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_ap.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_xmit.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_p2p.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_tdls.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_br_ext.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_iol.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_sreset.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/osdep_service.o
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/linux/os_intfs.o
/home/kali/Downloads/rtl8812au/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/home/kali/Downloads/rtl8812au/os_dep/osdep_service.c:1792:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1792 |         _rtw_memcpy(pnetdev->dev_addr, padapter->eeprompriv.mac_addr, ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/home/kali/Downloads/rtl8812au/os_dep/osdep_service.c:488:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  488 | void _rtw_memcpy(void* dst, void* src, u32 sz)
      |                  ~~~~~~^~~
/home/kali/Downloads/rtl8812au/os_dep/linux/os_intfs.c: In function ‘_rtw_drv_register_netdev’:
/home/kali/Downloads/rtl8812au/os_dep/linux/os_intfs.c:2446:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2446 |         _rtw_memcpy(pnetdev->dev_addr, padapter->eeprompriv.mac_addr, ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
In file included from /home/kali/Downloads/rtl8812au/include/drv_types.h:32,
                 from /home/kali/Downloads/rtl8812au/os_dep/linux/os_intfs.c:22:
/home/kali/Downloads/rtl8812au/include/osdep_service.h:167:35: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  167 | extern void     _rtw_memcpy(void* dec, void* sour, u32 sz);
      |                             ~~~~~~^~~
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/kali/Downloads/rtl8812au/os_dep/linux/rtw_android.o
  CC [M]  /home/kali/Downloads/rtl8812au/hal/hal_intf.o
/home/kali/Downloads/rtl8812au/os_dep/linux/rtw_android.c: In function ‘rtw_android_get_p2p_dev_addr’:
/home/kali/Downloads/rtl8812au/os_dep/linux/rtw_android.c:292:33: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  292 |         _rtw_memcpy(command, net->dev_addr, ETH_ALEN);
      |                              ~~~^~~~~~~~~~
In file included from /home/kali/Downloads/rtl8812au/include/drv_types.h:32,
                 from /home/kali/Downloads/rtl8812au/os_dep/linux/rtw_android.c:21:
/home/kali/Downloads/rtl8812au/include/osdep_service.h:167:46: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  167 | extern void     _rtw_memcpy(void* dec, void* sour, u32 sz);
      |                                        ~~~~~~^~~~
In file included from /usr/src/linux-headers-5.18.0-kali5-common/include/linux/export.h:45,
[... 84 lines cut ...]
  CC [M]  /home/kali/Downloads/rtl8812au/hal/OUTSRC/rtl8821a/HalHWImg8821A_BB.o
  CC [M]  /home/kali/Downloads/rtl8812au/hal/OUTSRC/rtl8821a/HalHWImg8821A_RF.o
  CC [M]  /home/kali/Downloads/rtl8812au/hal/OUTSRC/rtl8821a/HalHWImg8821A_TestChip_MAC.o
  CC [M]  /home/kali/Downloads/rtl8812au/hal/OUTSRC/rtl8821a/HalHWImg8821A_TestChip_BB.o
  CC [M]  /home/kali/Downloads/rtl8812au/hal/OUTSRC/rtl8821a/HalHWImg8821A_TestChip_RF.o
  CC [M]  /home/kali/Downloads/rtl8812au/hal/OUTSRC/rtl8821a/HalPhyRf_8821A.o
  CC [M]  /home/kali/Downloads/rtl8812au/hal/OUTSRC/rtl8821a/odm_RegConfig8821A.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_mp.o
  CC [M]  /home/kali/Downloads/rtl8812au/core/rtw_mp_ioctl.o
  LD [M]  /home/kali/Downloads/rtl8812au/8812au.o
  MODPOST /home/kali/Downloads/rtl8812au/Module.symvers
  CC [M]  /home/kali/Downloads/rtl8812au/8812au.mod.o
  LD [M]  /home/kali/Downloads/rtl8812au/8812au.ko
  BTF [M] /home/kali/Downloads/rtl8812au/8812au.ko
Skipping BTF generation for /home/kali/Downloads/rtl8812au/8812au.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-kali5-amd64'

                       ***Success***
***Module will be activated automatically at next reboot***

Do you wish to activate the module now? (y/n)y
***Module activated***
```

Any idea what this message `Skipping BTF generation for /home/kali/Downloads/rtl8812au/8812au.ko due to unavailability of vmlinux` means?

sxu0, 2022-09-23:
This did it for me originally, run with sudo: https://askubuntu.com/a/1404795

However, the issue arose once again for me, and a different fix is needed this time so I am all ears as well.

sxu0, 2022-09-23:
Try ignoring it as well. My install ended up going through smoothly despite the warning. Possibly relevant source: https://access.redhat.com/discussions/6970514