gnab/rtl8812au #253: UBSAN: array-index-out-of-bounds

ariccio, 2022-09-09:
Seen during early boot! (heh, I didn't even know it compiled with UBSan!)

```
[    6.419071] 8812au: loading out-of-tree module taints kernel.
[    6.419571] 8812au: module verification failed: signature and/or required key missing - tainting kernel
[    6.425361] ================================================================================
[    6.426296] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8812au/4.2.3/build/os_dep/linux/usb_intf.c:573:26
[    6.427407] index 3 is out of range for type 'int [3]'
[    6.427973] CPU: 6 PID: 389 Comm: systemd-modules Tainted: G           OE     5.15.0-47-generic #51-Ubuntu
[    6.427977] Hardware name: System manufacturer System Product Name/P8Z68-V PRO, BIOS 3603 11/09/2012
[    6.427979] Call Trace:
[    6.427982]  <TASK>
[    6.427985]  show_stack+0x52/0x5c
[    6.427995]  dump_stack_lvl+0x4a/0x63
[    6.428002]  dump_stack+0x10/0x16
[    6.428006]  ubsan_epilogue+0x9/0x49
[    6.428010]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[    6.428015]  ? _rtw_malloc+0x2d/0x33 [8812au]
[    6.428070]  rtw_drv_init+0x322/0x4cc [8812au]
[    6.428124]  usb_probe_interface+0xee/0x2b0
[    6.428129]  really_probe+0x222/0x420
[    6.428134]  __driver_probe_device+0x119/0x190
[    6.428138]  driver_probe_device+0x23/0xc0
[    6.428143]  __driver_attach+0xbd/0x1e0
[    6.428147]  ? __device_attach_driver+0x120/0x120
[    6.428151]  bus_for_each_dev+0x7e/0xd0
[    6.428155]  driver_attach+0x1e/0x30
[    6.428159]  bus_add_driver+0x139/0x200
[    6.428163]  driver_register+0x95/0x100
[    6.428167]  usb_register_driver+0x89/0x130
[    6.428171]  ? 0xffffffffc066a000
[    6.428174]  rtw_drv_entry+0x32/0x1000 [8812au]
[    6.428210]  do_one_initcall+0x48/0x1e0
[    6.428216]  ? kmem_cache_alloc_trace+0x19e/0x2e0
[    6.428224]  do_init_module+0x52/0x260
[    6.428228]  load_module+0xacd/0xbc0
[    6.428232]  __do_sys_finit_module+0xbf/0x120
[    6.428236]  __x64_sys_finit_module+0x18/0x20
[    6.428239]  do_syscall_64+0x5c/0xc0
[    6.428243]  ? syscall_exit_to_user_mode+0x27/0x50
[    6.428248]  ? __x64_sys_newfstatat+0x1c/0x30
[    6.428254]  ? do_syscall_64+0x69/0xc0
[    6.428257]  ? do_syscall_64+0x69/0xc0
[    6.428260]  entry_SYSCALL_64_after_hwframe+0x61/0xcb
[    6.428266] RIP: 0033:0x7fb7f5b20a3d
[    6.428270] Code: 5b 41 5c c3 66 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d c3 a3 0f 00 f7 d8 64 89 01 48
[    6.428273] RSP: 002b:00007ffd148e51f8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    6.428279] RAX: ffffffffffffffda RBX: 000055933c70b4b0 RCX: 00007fb7f5b20a3d
[    6.428281] RDX: 0000000000000000 RSI: 00007fb7f5c3f441 RDI: 0000000000000005
[    6.428284] RBP: 0000000000020000 R08: 0000000000000000 R09: 00007ffd148e5330
[    6.428286] R10: 0000000000000005 R11: 0000000000000246 R12: 00007fb7f5c3f441
[    6.428288] R13: 000055933c719700 R14: 00007fb7f5e846b9 R15: 000055933c7194a0
[    6.428292]  </TASK>
[    6.428294] ================================================================================
[    6.471707] usbcore: registered new interface driver rtl8812au

```

loganmancuso, 2022-10-19:
having this issue too, it worked on first dkms install but after reboot getting this error.

Fivetuple, 2023-02-17:
So am I,

kernel: [    5.757646] UBSAN: array-index-out-of-bounds in /var/lib/dkms/8812au/4.2.2/build/os_dep/linux/usb_intf.c:573:26