gnab/rtl8812au #255: too few arguments to function ‘cfg80211_ch_switch_notify’

thahn01, 2022-10-12:
CC [M]  /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/rtw_cfgvendor.o
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:478:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  478 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8812au/5.13.6/build/include/osdep_service_linux.h:98,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/include/osdep_service.h:54,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:17:
/usr/src/linux-headers-5.19.0-2-common/include/net/cfg80211.h:7967:6: note: declared here
 7967 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10415:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
10415 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10415:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10459:24: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  struct cfg80211_chan_def *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, struct cfg80211_chan_def *)’ [-Werror=incompatible-pointer-types]
10459 |         .get_channel = cfg80211_rtw_get_channel,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10459:24: note: (near initialization for ‘rtw_cfg80211_ops.get_channel’)
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.c:10695:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
10695 |         if (wdev->current_bss) {
      |                 ^~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.19.0-2-common/scripts/Makefile.build:254: /var/lib/dkms/rtl8812au/5.13.6/build/os_dep/linux/ioctl_cfg80211.o] Error 1

haodaweng, 2023-05-03:
@thahn01 have you got a solution for this?

thahn01, 2023-05-04:
I have switched PCs and the new one has a different wifi card.
Before that I think I have switched to
https://github.com/morrownr/8812au-20210629
Updates are a few months old and not a few years.